neonious/lowjs #1: N-API support

mhdawson, 2018-09-15:
I'm wondering there are plans to support [N-API](https://nodejs.org/api/n-api.html) for native modules.

neoniousTR, 2018-09-16:
We are planning on adding an API to add native modules to low.js. It will not the N-API, however. This we could add later on, of course, but other things are be more important.

I will write when the API for the native modules is implemented and documented.

mhdawson, 2018-09-17:
I'd be interested in understanding why something different from N-API?

neoniousTR, 2018-09-17:
Ah, there is a dedicated team for N-API? I understand your question now.

Looking more into this, I believe most napi functions can be quickly implemented by calling the correct DukTape equivalent. Should be just a few days of work.

So no, N-API would work well. But, it will have to be contributed if it shall be implemented very soon, because every hour we need for more basic things right now. Otherwise, it will be included some time later.

Thank you,

Thomas

mhdawson, 2018-09-17:
Thanks for the response, will raise for awareness in our next upcoming N-API meeting.  We are interested in seeing N-API across a number of implementation (it is already implemented on V8 and ChakraCore and @gabrielschulhof has been working a port for JerryScript) so that modules could be written once to run across the different implementations.  We are already pretty busy with current activities, but at least will mention in case there is somebody with an interest in helping out.

natevw, 2018-11-01:
In the meantime is there a lower-level way to integrate native functionality? For example my https://github.com/natevw/pi-spi has a very small `nan` surface and perhaps could be ported to low.js in a different way?

neoniousTR, 2018-11-02:
If you know how, it is quite easy already.
I have to create a custom native module for a project this week anyhow, so I will make sure I create a template and add it to the repo.

neoniousTR, 2018-11-08:
Well, thank you for your patientance!

See examples/low_with_native_module for an example on how to add a custom native module (direct on DukTape functions, not N-API).

Closing issue for now!

closed 2018-11-08 by neoniousTR