neonious/lowjs #16: TypeError: cannot find module 'lib:init'

Stradivario, 2018-10-10:
Trying to run @rxdi/core with low.js and i am facing the following error:

```
TypeError: cannot find module 'lib:init'
    at [anon] (src/low_module.cpp:226) internal
Segmentation fault (core dumped)
```

https://github.com/rxdi/core/tree/master/dmodule

Btw @rxdi/core can be found on decentralized space:

https://ipfs.infura.io/ipfs/QmP9n7m1UWkFn2hqt7mnfQYnBtxmFyyvKZHN7hgngQb1gM

Here is a module which is working on regular Typescript compilation to commonjs module.

Any ideas ?

neoniousTR, 2018-10-10:
Interesting. Would like to reproduce your problem. What do I have to do exactly to get your error?

neoniousTR, 2018-10-29:
After a bit of testing, I think you started low without having the ../lib directory built. So it seems to be a user error.

The segmentation fault should not have been, however, fixed in master.

closed 2018-10-29 by neoniousTR

Stradivario, 2018-10-30:
@neoniousTR i am really sorry that i didn't have time to write you since i introduce the issue but i am on 3 projects at the same time and i really don't have time even to sleep.Maybe i somehow copy only binary instead copy the whole lib yeah sounds reasonable! Because now when i test it i am having another error which is different.

Normally i start my project like this

```
node ./src/main.js
```

when i try to run it with low js it throw the following error:

```
./low ~/Desktop/work/my-project/src/main.js
```

```
SyntaxError: empty expression not allowed (line 22)
    at [anon] (/home/rampage/Desktop/work/my-project/src/main.js:22) internal
    at [anon] (deps/duktape/src-low/duktape.c:69865) internal
```

Do you know what is happening ?