neonious/lowjs #17: Docs/examples for using packages (eg, websockets)

BrendanFDMoore, 2018-10-12:
Hi there,

I've been trying to figure out how I'd go about using a package/module such as `ws` or `socket.io-client` to connect to a WebSocket I run on another server. While the documentation does indicate that `node_modules` is not supports, the web server example does `require` several modules. How can I do that for another module?

Basically I'm trying to set up my device (Neonious One) as a small client for receiving messages via socket to set various pins high or low to control relays etc. I'd run the socket server and the UI app elsewhere.

Thanks!

neoniousTR, 2018-10-13:
Hi!

Did you buy the neonious one, or do you have it from JSConf? Please keep in mind that the JSConf version runs of an old software version and has limited functionality, and this will not work.

If you bought the neonious one, we will check why ws does not work yet, and fix this.

Looking forward to your response.

Thomas

BrendanFDMoore, 2018-10-14:
Hi Thomas

I did get the N1 at JSConf, and thank you for the generous loot!

Ok, that kind of makes sense. Would you expect that updating to the latest software would enable this? I'm going to try to run the update and see if things change, but previously the Package Manager search results were a bit sparse. Is this because compatibility here requires you to do per-package work?

Cheers :beers:

neoniousTR, 2018-10-15:
OK I understand now.

The neonious one from JSConf does not get updates. To get updates, you need to buy the neonious one, or pay the equivalent to get the update for the neonious one you already have (you do not have to pay shipping then). See the TPManual for this.

The package manager in this version only included a few packages because it was only a preview of the future package manager.

Regarding websockets: We took a look at the ws package of npm. Not much is missing to get this working with low.js. Our plan is to get this done mid November.
You will then of course be able to use this with low.js for PC. You can only use the finished version on the neonious one itself when paying though (see above)

Sorry, and thank you for your understanding.

neoniousTR, 2018-10-28:
ws module is now working well with low.js. Check the webserver example in the git master, it uses WebSockets now.

lamanabie, 2018-11-07:
@neoniousTR Hi, I am not sure how you implemented parsing http in lowjs, but you can check http-parser-js, that is pure javascript http parser and have same api as native http parser in nodejs,  https://github.com/creationix/http-parser-js

neoniousTR, 2018-11-08:
You are right. Will add the client part as soon as it is on the top of our list. it is just a few lines of code, so no need to change the complete parser (which is well optimized and working well).

lll000111, 2019-01-26:
> Only server part of ws module is working, but using client ws does not

We too need the ws client - our code is both WS server and client... (cc @winpuc)

neoniousTR, 2019-01-28:
The current pipeline here is as follows:

- Get lowsync API and UI in IDE for npm modules working (works in the lab already, really is fun!)
- Small revamp of lowsync (better monitor and so on)

After that we could look into WebSockets.
If this is not fast enough, you can take a look at low.js yourself. The code for this is ESP32-agnostic so it is part of the Open Source code.

lll000111, 2019-01-28:
Just to clarify &mdash; is this about *native* WebSocket support for low.js? I thought we already dealt with the "ws" package some time ago. That package provides both client and server websocket functions and is what we use so we don't need the functionality natively (node.js does not have it either). Or does the "ws" package not fully work yet?

neoniousTR, 2019-01-28:
No, there is no native WS, as Node.JS dos not have it either.
"ws" package works, but it requires a fully functioning http/https including "upgrade" event. And this we need to implement on the server side.

lll000111, 2019-01-28:
Now I'm more confused.

> And this we need to implement on the server side.

So a WebSocket *server* does not work yet? But a client does? Or vice versa? Or none? (All while using "ws".)

neoniousTR, 2019-01-28:
You are right... : Server works, client does not yet.

neoniousTR, 2019-05-23:
Marking this an enhancement, as WebSocket clients is a feature to be added, not a bug to be fixed.

neoniousTR, 2019-06-19:
low.js now supports WebSockets clients! Please read our newsletter for more details: https://www.neonious-basics.com/index.php/2019/06/19/newsletter-june-19th-2019/
Or, just try it out!

closed 2019-06-19 by neoniousTR