neonious/lowjs #18: Documentation: How to run mocha tests using low binary instead of node

lll000111, 2018-10-14:
Before I run code on the device I would like to use the platform binary or low.js to run the transpiled code on the development system. Specifically and as a use case example, I want to run the existing mocha tests (used for node.js and browser thus far).

This does not seem to be trivial, I can't find a way to run mocha tests on low. I would appreciate this added to documentation. Automated tests are a must, how would one use one of  the most widely used tets runners with low.js?

neoniousTR, 2018-11-03:
Sorry for the late answer!

Well, when low.js is done it should just work like Node. If it is not, something is simply not implemented yet.

Another thing which could be in the way is that low.js is only ES5 (ES6 seamlessly with automatic transpilation when syncing with ESP32), while some modules in npm are ES6. For use cases like this (tests on PC), we are planning on adding optional automatic transpilation with babel to the low binary. 

I would say, tell us what you already tried, but I guess you have an idea now, because you opened issue 24. Closing because of this.

lll000111, 2018-11-07:
(Please reopen)

Possibly relevant: https://github.com/mochajs/mocha/issues/956

neoniousTR, 2018-11-07:
Trying to get an example going with mocha is high on our list. I guess this will happen this month. I will report here.

lll000111, 2018-11-07:
I added my observations in https://github.com/mochajs/mocha/issues/956#issuecomment-436758656 - they already support running mocha from JS in the browser, strange that the same steps don't work from node.js. I'm too lazy to now debug the `mocha.js` file though.

lll000111, 2018-11-08:
The current stopper is npm dependency "supports-color" of mocha &mdash; because it has a `let` statement in line 7 of its index.js file.

Duktape and therefore low.js already accepts `const` (I think Duktape treats it as `var`), `let` should be possible on the same (preliminary, incomplete) level.

Reference: https://github.com/svaarala/duktape/issues/2000

neoniousTR, 2018-11-08:
No, let is not yet supported in DukTape.

But, as I wrote: "For use cases like this (tests on PC), we are planning on adding optional automatic transpilation with babel to the low binary."

Hopefully this will be done this month.

neoniousTR, 2019-01-28:
Just FYI: The current pipeline here is as follows:

- Get lowsync API and UI in IDE for npm modules working (works in the lab already, really is fun!)
- Small revamp of lowsync (better monitor and so on)
- WebSockets Server

After that we could look into mocha.

lll000111, 2019-01-28:
You can make mocha a lower priority before any other blockers, I can code-write my way around that.

neoniousTR, 2019-05-23:
Will not be done in foreseable future.

closed 2019-05-23 by neoniousTR