neonious/lowjs #24: Attempting to load mocha.js: Segmentation fault (Linux)

lll000111, 2018-11-02:
low.js version lowjs-linux-x86_64-20181102

mocha is globally installed. I call

    const m = require('/usr/lib/node_modules/mocha/lib/mocha.js')

and get "Segmentation fault" and nothing else. The file exists.

At least a useful error message would be nice.

The source code for the file is available here: https://github.com/mochajs/mocha/blob/v5.2.0/lib/mocha.js

neoniousTR, 2018-11-03:
Can you retry with the version of tonight (20181104)? Will be available in 2 hours. Fixed two things:

- require with absolute paths now work correctly
- diff module is resolved correctly (main file specified in package.json as directory)

Thank you

lll000111, 2018-11-05:
New version downloaded just now, then

```
mha@mint-virtual-machine ~ $ ~/lowjs-linux-x86_64-20181105/bin/low 
> const m = require('/usr/lib/node_modules/mocha/lib/mocha.js')
Segmentation fault
```

neoniousTR, 2018-11-05:
I can reproduce the problem. Fixing it ASAP.

neoniousTR, 2018-11-06:
The segfault should be fixed in the version 181106 which will be available in a few hours. Looking forward to your progress report! Thank you!

lll000111, 2018-11-06:
Please reopen #18. I still think it should be documented how to run tests using the most commonly used test frameworks.

There is no crash any more, but I still have no idea how to get this running. I tried doing what I do in the browser tests but quickly ran into

```
> mocha.run()
TypeError: cannot read property 'search' of undefined
    at [anon] (deps/duktape/src-low/duktape.c:56562) internal
    at [anon] (/usr/lib/node_modules/mocha/mocha.js:149) strict
    at global (repl:1) preventsyield
    at [anon] () native strict preventsyield
    at [anon] (lib:vm:17) strict
    at defaultEval (lib:repl:260) strict
    at onLine (lib:repl:511) strict
    at emit (lib:events:158) strict
    at [anon] (lib:readline:231) strict
    at [anon] (lib:readline:530) strict
    at [anon] (lib:readline:773) strict
    [...]
```

I don't think it's worth solving this problem in isolation, there will just be another one.

Another issue are tests using some ES2015+ syntax. Mocha run from the command line has a hook for Babel option (although I'm not sure if that is for test files too, or just code files, or both).

neoniousTR, 2018-11-07:
FYI about ES2015+ syntax:

low can be used on Linux devices as very lean environment, or it can just be used on Linux devices to test things which will later run on ESP32. For the latter we will add on-the-fly babel translation to the low binary in the next 2 weeks.

closed 2018-11-07 by neoniousTR