neonious/lowjs #29: IDE: "Upload directory" does not work

lll000111, 2018-11-08:
I started the IDE provided by the NEONIOUS ONE and tried to "upload directory", using my local lib/ directory with transpiled files. Nothing happened except his:

[image]

lll000111, 2018-11-08:
SOLVED - but that just means there's another problem:

Turns out the session had expired. However, I only found out after.. "doing stuff", an overlay popped up telling me to reload the page.

1) The IDE should tell me as soon as it happens (that the session has timed out)

2) The IDE session should not time out

lll000111, 2018-11-08:
Another problem: The upload is very slooooooooooooooooow. The network activity gadget of the IDE (probably correctly) shows 100-200 Bytes/sec (no typo). I'm connected through Ethernet (wire), WiFi is off on the device. Earlier I had trouble loading the IDE from the (freshly started) device, after USB out/USB in (to restart) it worked, but now that slow upload.

lll000111, 2018-11-08:
Actually, I cannot reach the device any more

lll000111, 2018-11-08:
New attempt using  `lowrmt` instead of the IDE in the browser. Result:

```
mha@mint-virtual-machine ~ $ lowrmt sync
The password for your device was not found in the configuration.
? Please enter the password of the device. [hidden]
? Save new password into configuration? No
Syncing to remote core-js/client/core.min.js
+File "/home/mha/neo/core-js/client/core.min.js" -> "/core-js/client/core.min.js"
An error has occured: An unexpected error occured while communicating with the remote. RequestError: Error: socket hang up
```

neoniousTR, 2018-11-08:
Well, I must say, you are most one of the early users of our software.
Sorry for this. Looking into it tonight.

lll000111, 2018-11-08:
No worries, it is what it is.

 ¯\\\_(ツ)\_/¯

lll000111, 2018-11-10:
If my router is right it's 30:AE:A4:60:30:CF

neoniousTR, 2018-11-13:
Have to postpone to 10 PM today. Everything is working, but with the new version SPI and UART have some changes, and need to update the documentation before enabling the update. Can only do this tonight.

neoniousTR, 2018-11-14:
OK, it took longer than I thought. Please try out the update which you should be able to install in the IDE.

lll000111, 2018-11-14:
If I remember that right from the previous update I should get a button in the upper right corner of the IDE? Right now I don't see anything.

neoniousTR, 2018-11-14:
Your device (with the MAC above) connected to our update server exactly 2 min before your post. Maybe it took to long? Everything looks correct, can you retry?

lll000111, 2018-11-14:
I just retried and saw nothing.

neoniousTR, 2018-11-14:
OK, we need to test this more in the future.
Had the  same problem with a board here, and it should be fixed.

lll000111, 2018-11-14:
More than half an hour has passed and the red light keeps blinking. i would like to turn it off because lunch is waiting... last time it was all updated after a minute or two.

neoniousTR, 2018-11-14:
Well, turn it on after lunch again, please, maybe it's working now. If not, we will figure it out..
I am keeping the update log of the server open to see what's happening.

Background Info:
ESP-IDF itself really is not stable in the master branch. It's a big software package and is being updated fast. We merged changes, and now the connection to the update server is flacky (mbedtls part of ESP-IDF is the problem). The problem started with our software version 1.2.0. We worked around the first API call from server side (Is Update available?) and I think the other version we will be able to work around, too (It seems like the last work round from . few hours ago is not working yet everywhere).

Conclusion:
We will only deploy updates after testing of all systems. For this we will add unit tests as one of the very next steps.

lll000111, 2018-11-14:
I turned it back on and it just keeps blinking (red), homepage still loads and says "neonious one is updating". It's been almost an hour.

neoniousTR, 2018-11-14:
Seeing the requests. Keep it on please, looking into it.

neoniousTR, 2018-11-14:
Is a different problem now (first problem was disconnect right after connect, now it stalls). Are you using the Ethernet port? In this case, this could be the same problem you had with the IDE upload. Now added a rate limiter, we will see if it works soon. Upload now takes 20 Min, but if it works then, it was worth it.

neoniousTR, 2018-11-14:
Yeah well, as we are fixing the Ethernet port now we should have maybe thought about switching to Wifi before updating :-) What a drama.

Just for testing purpose: If it is still updating, can you quickly reboot it? Just added some logging features and it would be great if it reconnected.

lll000111, 2018-11-14:
Done.

neoniousTR, 2018-11-14:
OK. This is no use. It just stalls. You can turn it off.

Let me do several tests this evening. Scenarios:

(1) The problem is the Ethernet port (which most probably is the case because I could update at least with the changes I made - it was two different bugs in ESP-IDF). In this case I hopefully can find a configuration which works.

(2) If this problem is the Ethernet port, but this is unsolvable, there are two options:

I think Set Settings with lowrmt works in Updater mode. You could enable Wifi and disable Ethernet. I can tell you whether your settings change worked by looking in the log. (You could try this now, too, but only if you have spare time, please).

I think you have a second board? You could update it via Wifi and we would send a replacement board. But for this, let me check first.

(3) In the worst case scenario the problem still has nothing to do with Ethernet, lets talk about that after the tests. (Good that only a handfew of boards were updated to 1.2.0 by other people yet)

Thomas

lll000111, 2018-11-14:
The red light stopped and the IDE loads, is this the new or the old version now?

neoniousTR, 2018-11-14:
Does it say version 2018-11-13 on the login page? If yes, congrats! The rate limiter did work after all and it was the Ethernet port.. I guess this was a warning for us to start doing unit tests :-)

lll000111, 2018-11-14:
[image]

lll000111, 2018-11-14:
`lowrmt` sync works so far, lots of files already synced.

neoniousTR, 2018-11-14:
We know of some problems with lowrmt, might work far better when the update we are testing is deployed. But good if things are working for now. Closing this issue, but of course ready for new ones :-)

lll000111, 2018-11-21:
*UPDATE*

I'm using the IDE "upload directory" feature right now, device software version 19 November. The speed issue still exists, uploading 250 kByte takes _ages_, upload speed is in the single digit kByte/s.

lll000111, 2018-11-21:
Right now I cannot try anything at all, because all I get is a stack overflow after adding some code (just loading some of our modules, I'm not even doing anything with them yet). So all I get is a broken IDE, expired session message, and I have to try to somehow quickly disable those lines of code before I get logged out again.

Honestly,. I think the IDE is a nice touch but nothing but a toy in the end (sorry for being quite frank). I think development should be off-device, on the PC, with remote debugging from an actual IDE.

Orthogonal issue: If there is an issue the device should not attempt to run the code again or I end up with an endless restart loop. I guess that's the IDE doing it though.

neoniousTR, 2018-11-21:
How do you know it's a stack overflow? Does it say so?

Let me think about this. Hopefully I know how to proceed sometime tonight.

Thomas

neonious GmbH

Münsterstr. 246 / 40470 Düsseldorf / Germany

T   +49 211 9241 8187
F   +49 211 9241 8172

www.neonious.com <http://www.neonious.com>

Managing Director & CEO: Thomas Rogg
Commercial Register: Amtsgericht Düsseldorf, HRB 83086

lll000111, 2018-11-21:
While the IDE still is up for a few seconds I can see it, and the final popup message on the automatically loaded login page says

An fatal error has occured in the program:

Error: stack full (ask neonious when this is more flexible)

Program cannot continue.

neoniousTR, 2018-11-22:
First, sorry for the problems. Yes, we really are at an early stage.

What the problem is:

(1) Every require adds around 20K to the CPU thread stack (not the JavaScript stack), and when they are nestled, the CPU thread stack is at it limit quickly. This is different from calling JavaScript recursivly. When that is done, the JavaScript stack builds up, but it does not use additional bytes from the underlying CPU thread stack, so here there is no limit other than the 4 MB heap.

(2) There is an exception when the CPU thread stack is at it limit. But it seems like it is called too late. So yes, the program halts. But also after that, the OS crashes, because we have a real stack overflow.

(3) Because of the reboot of the MC the session of the IDE gets invalid.

(4) Also, on the other side: Writing data to the flash is not fast. With lowrmt it takes even longer, because currently it builds a new HTTPs stream for every file, and the handshake takes a second. The fastest right now will be HTTP over the IDE. So, please try that out.

What the solution is:

(1) There is an update available now which allows require() chains limited only by the heap. This is done through a unorthodox operation of copying the stack to the heap when the stack grows too much. Looking through the code I think all cases are handled. It seems to work, please try it out.

(2) There will be an update of lowrmt next week, which actually makes the sync usable well. Then, I think, you will no longer need the IDE.

(3) Also, we have on the list to make low.js save a crash report in a file when the OS crashes + skip auto-start if a crash happens within 5 min of boot. Sorry if you will have to resync once.

Thomas

lll000111, 2018-11-22:
Using the IDE right now (would there be a difference in how this is handled if I used the CLI tool?). I'm having trouble with 

    Object.values is undefined

I uploaded a folder lib/ with ES6 code files to the top level. Within the src/ folder there is a file that loads modules from there using (for example) `const CoreTypes = require('../lib/core-types.js');` Loading the module worked but `Object.values` was missing, was there no ES6 transpilation?

Next, in one of those included files I create a method

```js
module.exports.test = () => {
    return new Set([1, 22, 3]);
};

// Prevent further parsing of the rest of the file
return;
```

From a /test URL path request function in index.js I then create string output `const str = [...ThatFile.test()];` and send it to the browser. I get 

    undefined not callable (property 'slice' of [object Object])

I also tried it directly in index.js.

It seems to me that there is transpilation, otherwise I would have seen a syntax error, but it does not actually work? Using `String(new Set())` results in `[object Object]`, it should be `[object Set]`.

So... how does the automatic transpilation actually work?

neoniousTR, 2018-11-22:
Good, the stack compression works. (Note that I've noticed a crash every 
dozend tries... But this will be gone in a few days, tooo).

Transpilation is done by TypeScript if using Webbrowser. We are using 
TypeScript, because some day we can then just enable that people can use 
TypeScript syntax and not only ES6 in the webbrowser.

lowrmt is using babel. Not sure if this was a good decision, we might 
change to TypeScript with lowrmt, too, so we have the same behaviour

We should allow the user to look into the transpiled files, but 
unfortunaly that is not developed yet.

Can you isolate the problems into simple examples with a few modules and 
few lines of code? If so, we will fix them as fast as we fixed the other 
things.

neoniousTR, 2018-11-22:
Note: Object.values() is not even ES6, it's even newer: 
https://developer.mozilla.org/de/docs/Web/JavaScript/Reference/Global_Objects/Object/values

I would suggest not to use it.

neoniousTR, 2018-11-22:
Note:
The difference in reporting (see below) is now on my list. For 
everything else, please give me more info.

Thomass-MacBook-Pro:~ thomas$ low
 > String(new Set())
'[object Object]'
 >
(To exit, press ^C again or type .exit)
 >
Thomass-MacBook-Pro:~ thomas$ node
 > String(new Set())
'[object Set]'
 >

lll000111, 2018-11-26:
> or both TypeScript compiler-based. 

Please, Babel, that's the point... with Babel you get TS too, with TS you get only TS. Not everybody writes Typescript, and knowing both Flow and TS pretty well after years of using and reading issues on Github and reporting them and answering many I have to say there is way too much hype from people who barely know both. Both TS and Flow have lots and lots of issues. Flow has the advantage of not changing the actual Javascript (just remove the types and it's Es 2015+), with TS that's not always true.

Also, as I describe above, in the 1st point, TS here has a bug &mdash; definitely a bug, since it produces non-functioning code (if it can't do something it could/should produce an error, not silently produce code that attempts to use a function that does not exist). Babel works.

**Note:** The issue with `Set` is not (just) the string description, it is this:

    [...(new Set([1,22,3]))]

This is perfectly valid ES 2015 code that produces an array `[1 ,2, 3]` from a `Set`.

[TypeScript Playground](http://www.typescriptlang.org/play/#src=%5B...(new%20Set(%5B1%2C%2022%2C%203%5D))%5D%0D%0A): The code produced by TS does not work!

[Babel version](https://babeljs.io/repl#?babili=false&browsers=&build=&builtIns=false&spec=false&loose=false&code_lz=NoOjAoDsFMHcAIDK0Au5gEYA0AmHWBmAXQEoSiAoIA&debug=false&forceAllTransforms=false&shippedProposals=false&circleciRepo=&evaluate=false&fileSize=false&timeTravel=false&sourceType=module&lineWrap=true&presets=es2015%2Ces2017%2Cstage-2%2Cstage-3&prettier=true&targets=&version=7.0.0-rc.1&envVersion=1.6.2): This works! (conversion to ES5)

lll000111, 2018-11-26:
> Regarding your question:
> Babel in lowrmt runs on PC before sycing.
> TypeScript in IDE runs in browser. It is in the ~10 MB which is loaded when starting the IDE.

No, my question was about the *runtime code*. Polyfills and "support code" used to polyfill ES 2015+ features after down-compilation. Those libs take up space. `core-js` and `@babel/runtime` when you use Babel.

neoniousTR, 2018-11-26:
Those polyfills are in init.js which runs before the program runs. Yes, it needs some space, but as you can see we only use select polyfills right now. It fits quite well.

You are right.. [...(new Set([1,22,3]))] produces non-working code if downlevelIteration is false. It is true in the newest update, so this is not a problem. This produces working code if Symbol.iterator for Set exists (which it does now). But the Babel version with Array.from is nicer, true.

I understand Babel is the better choice. We will look at that, but it's lower prio.

lll000111, 2018-11-27:
The firmware has a date of 26 November, but I don't think that's the one you meant since it still says

> An fatal error has occured in the program:

> TypeError: undefined not callable (property 'slice' of [object Object])

> Program cannot continue.

which is the TypeScript bug mentioned above and you said that should not happen with the mentioned TS compiler flag set. Also, latest changelog entry is 22 November.

neoniousTR, 2018-11-27:
Try triggering a retranspilation bei adding a new line to the file. The update is correct we will update changelog tonight.

Thomas'

lll000111, 2018-11-28:
What I did was, from the IDE,

- Upload the lib/ folder (ES 2015+ code)
- Add one line to load one of the modules (a pretty simple one)
>  const ObjectToMicrodata = require('../lib/object-recipes.js');

Model popup in the IDE:

> Program is paused, but cannot locate the line it was paused on!

Error:

```
Error: stack full (ask neonious when this is more flexible);
             Recipe: {"name":"Access","rule":[{"isId":true,"itemprop":"object","referenceTo":{"_itp":[],"_keys":[],"_values":["*"]}},{"idReferenceTo":{"_itp":[],"_keys":[],"_values":["Person"]},"itemprop":"person","list":"orderedByONE"},{"idReferenceTo":{"_itp":[],"_keys":[],"_values":["Group"]},"itemprop":"group","list":"orderedByONE"}],"type":"Recipe"}
             at [anon] internal
             at [anon] (/lib/type-checks.js:185) strict preventsyield
             at every () native strict preventsyield
             at ensureRecipeRule (/lib/type-checks.js:185) strict
             at [anon] (/lib/type-checks.js:252) strict preventsyield
             at map () native strict preventsyield
             at [anon] (/lib/type-checks.js:252) strict
             at [anon] (/lib/object-recipes.js:67) strict
             at [anon] (/lib/object-recipes.js:126) strict preventsyield
             at require () native strict preventsyield
             at [anon] (/src/index.js:65) preventsyield
             [...]
```

I then removed the type-checks.js module and tried again, works. So I added a few more types, same as object-recipes.js already does ("types" means JS objects of the kind `{type: "MyType", someProp: "FooBar"}`, each type describing a different object with a "type" property, so really simple stuff, just very basic JS objects).

Now this error:

```
Error: stack full (ask neonious when this is more flexible)
             at [anon] internal
             at [anon] (/src/register-types.js:497) strict preventsyield
             at some () native strict preventsyield
             at [anon] (/src/register-types.js:497) strict preventsyield
             at reduce () native strict preventsyield
             at [anon] (/src/register-types.js:501) strict preventsyield
             at require () native strict preventsyield
             at [anon] (/src/index.js:66) preventsyield
```

In the IDE highlighted is the third line of

```js
// "module.exports" is line #500
module.exports.VERSIONED_OBJECTS = RECIPES.reduce(
    (arr/*: Array<string>*/, recipe/*: RecipeObj*/)/*: Array<string>*/ => {
        if (recipe.rule.some((rule/*: RecipeRule*/)/*: boolean*/ => rule.isId === true)) {
            arr.push(recipe.name);
        }

        return arr;
    },
    []
);
```

Not sure if those line numbers are correct?

lll000111, 2018-11-28:
So I just tried **`lowrmt`* and its Babel (6) transformation. Result:

```
TypeError: undefined not callable (property [Symbol wellknown '?Symbol.iterator?'] of [object Array])
             at [anon] internal
             at [anon] (/lib/object-recipes.js:34) strict preventsyield
             at require () native strict preventsyield
             at [anon] (/src/index.js:65) preventsyield
```

I think the line numbers are of the transformed file, which I cannot see. Whatever this is, it shoudl have been transformed.

neoniousTR, 2018-11-29:
Both fixed. Testing, because a few other things were added with the new version. Will write when update is available.

Thank you for your patience (hopefully)..

lll000111, 2018-12-03:
I don't know how to debug the latest issue, because the exact same code behaves differently in different contexts.

Trivial helper `isString` is defined as

```
function isString(thing) {
  return typeof thing === 'string';
}
```

The simplified test code line:

```
(['string'].every(type` => isString(type)) ? 'TRUE' : 'FALSE')
```

### Case 1

To the demo `http.createServer(...` function in the demo index.js file I added

```
    } else if (req.url === '/test1') {
        res.writeHead(200, { 'Content-Type': 'text/plain' });
        res.end((['string'].every(type => isString(type)) ? 'TRUE' : 'FALSE'));
    }
```

When I call /test1 I get TRUE.

### Case 2

I have a lengthy function `function ensureRecipeRule(thing; any)` with lots of "if" statements checking various properties on a given object to ensure that the object is what I expect, so it is a runtime type refinement check. One such test is

```
  if (thing.referenceTo !== undefined && (!(thing.referenceTo instanceof Set) || ![...thing.referenceTo].every(type => isString(type)))) {
    throw new TypeError(
      `Rule #${index}: optional "referenceTo" must be a Set of ONE object type strings or "*"\n ` + 
      'referenceTo: ' + [...thing.referenceTo].join(' : ') + '\n' +
      'isSet: ' + (thing.referenceTo instanceof Set ? 'TRUE' : 'FALSE') + '\n ' +
      'isEvery: ' + (['string'].every(type => isString(type)) ? 'TRUE' : 'FALSE')
    );
  }

```

This code is not run in response to a URL request but during webserver startup. I simply `require` one of our modules, which loads a bunch of objects, each one has to pass through that type check function that checks the properties.

Here, `isEvery` always is `FALSE`.

Console output:

```
10:14:54 AM] TypeError: Rule #4: optional "referenceTo" must be a Set of ONE object type strings or "*"
              referenceTo: *
              isSet: TRUE
              isEvery: FALSE;
              Recipe: {"name":"PlanEnd","rule":[{"itemprop":"plan","regexp":/^[0-9a-f]{64}$/},{"itemprop":"startTime","valueType":"number"},{"itemprop":"endTime","valueType":"number"},{"itemprop":"child","list":"orderedByONE","referenceTo":{"_itp":[],"_keys":[],"_values":["Plan"]}},{"itemprop":"object","list":"orderedByONE","rule":[{"itemprop":"reference","referenceTo":{"_itp":[],"_keys":[],"_values":["*"]}},{"itemprop":"type"},{"itemprop":"status","regexp":/^(new|exists)$/},{"itemprop":"idHash","optional":true,"regexp":/^[0-9a-f]{64}$/},{"itemprop":"timestamp","optional":true,"valueType":"number"}]},{"itemprop":"clob","list":"orderedByONE","rule":[{"clobLink":true,"itemprop":"hash"},{"itemprop":"status","regexp":/^(new|exists)$/}]},{"itemprop":"blob","list":"orderedByONE","rule":[{"blobLink":true,"itemprop":"hash"},{"itemprop":"status","regexp":/^(new|exists)$/}]},{"itemprop":"returns","list":"orderedByApp","regexp":/^[0-9a-f]{64}$/}],"type":"Recipe"}
              at ensureRecipeRule (/lib/type-checks.js:189) strict
              at [anon] (/lib/type-checks.js:209) strict preventsyield
              at every () native strict preventsyield
              at ensureRecipeRule (/lib/type-checks.js:209) strict
              at [anon] (/lib/type-checks.js:255) strict preventsyield
              at map () native strict preventsyield
              at [anon] (/lib/type-checks.js:255) strict
              at [anon] (/lib/object-recipes.js:51) strict
              at [anon] (/lib/object-recipes.js:147) strict preventsyield
```

The output for "referenceTo" properties, which are ES2015 Set objects, does not show what's inside them, but it's just `*` as my manually created debug output in the first three lines shows.

lll000111, 2018-12-03:
I mixed using `lowrmt` and the IDE, so I guess that mixes TypeScript and Babel output. I'm trying again with just `lowrmt` and deleted the uploaded folder in the IDE.

lll000111, 2018-12-03:
When I do *everything* with `lowrmt`, including re-uploading index.js, I get

```
[11:36:25 AM] ReferenceError: identifier 'regeneratorRuntime' undefined
              at [anon] internal
              at [anon] (/src/index.js:4) strict
              at [anon] (/src/index.js:34) strict preventsyield
```

lll000111, 2018-12-10:
Closing this issue and using small individual issues instead.

closed 2018-12-10 by lll000111