neonious/lowjs #32: Map and Set objects are not identified correctly

lll000111, 2018-12-04:
Extracted from https://github.com/neonious/lowjs/issues/29#issuecomment-440972157

Just FYI, there is a similar issue in React Native's runtime: https://github.com/facebook/react-native/issues/19594

```
// Common stuff that works:
Object.prototype.toString.call(new Array())
// [Object Array]
Object.prototype.toString.call(new Date())
// [Object Date]

// Map and Set don't work:
Object.prototype.toString.call(new Set())
String(new Set())
// [Object object] but should be [Object Set]

Object.prototype.toString.call(new Map())
String(new Map())
// [Object object] but should be [Object Map]
```

This self-identification of constructors in JS is used quite frequently. If it is incorrect hacks have to be used.

neoniousTR, 2019-01-16:
Will change to Symbol.toStringTag when we have time.
Till then, it works perfectly the way it is. Just keeping this issue open to not forget.

In comment to commit:

Is this the best option, instead of setting Symbol.toStringTag for each object type and leaving Object.prototype.toString unchanged?

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Symbol/toStringTag

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Map/@@toStringTag

https://tc39.github.io/ecma262/#sec-object.prototype.tostring (item 15 "Let tag be ? Get(O, @@toStringTag).")

lll000111, 2019-01-30:
Just to update what needs to be done:

    Map.prototype[Symbol.toStringTag] = 'Map'

http://2ality.com/2015/09/well-known-symbols-es6.html#overriding-the-default-tostring-tag

commit 8a676afd (2019-01-15): fixed #32
commit 1963419d (2019-02-03): fixes #32
closed 2019-02-03 by neoniousTR