neonious/lowjs #33: crypto: SHA-256 hashes in hex notation are not always 64 characters

lll000111, 2018-12-09:
Apparently hex values like "0a" are written into the final SHA-256 hash as just "a" when the hash string is created from its components.

```
> mha@mint-virtual-machine ~/neo $ node
> const crypto = require('crypto')
undefined
> crypto.createHash('sha256').update("strin", 'utf8').digest('hex').length
64
> crypto.createHash('sha256').update("strin", 'utf8').digest('hex')
'98db433e68d56413cb026680937df70118070c105d0c8a74f4b6f8ec5ea140ee'
```

vs.

```
> var crypto = require('crypto')
undefined
> crypto.createHash('sha256').update("strin", 'utf8').digest('hex').length
62
> crypto.createHash('sha256').update("strin", 'utf8').digest('hex')
'98db433e68d56413cb026680937df7011807c105dc8a74f4b6f8ec5ea140ee'
```

neoniousTR, 2018-12-10:
fixed in git, will be in download version tomorrow morning
will create new package for esp32 + neonious when at least one other commit is fixed tomorrow night.

neoniousTR, 2018-12-10:
BTW: thank you for the tip with the github link. we will add that

commit 36a09fd2 (2018-12-10): fixed #33

also some documentation fixes
closed 2018-12-10 by neoniousTR