neonious/lowjs #35: DukTape error: "unsupported"

lll000111, 2018-12-10:
Babel produced this code, all the function at the top are Babel helpers:

```js
'use strict';

// BABEL INTERNAL HELPERS

function _typeof(obj) { if (typeof Symbol === "function" && typeof Symbol.iterator === "symbol") { _typeof = function _typeof(obj) { return typeof obj; }; } else { _typeof = function _typeof(obj) { return obj && typeof Symbol === "function" && obj.constructor === Symbol && obj !== Symbol.prototype ? "symbol" : typeof obj; }; } return _typeof(obj); }
function _classCallCheck(instance, Constructor) { if (!(instance instanceof Constructor)) { throw new TypeError("Cannot call a class as a function"); } }
function _possibleConstructorReturn(self, call) { if (call && (_typeof(call) === "object" || typeof call === "function")) { return call; } return _assertThisInitialized(self); }
function _assertThisInitialized(self) { if (self === void 0) { throw new ReferenceError("this hasn't been initialised - super() hasn't been called"); } return self; }
function _inherits(subClass, superClass) { if (typeof superClass !== "function" && superClass !== null) { throw new TypeError("Super expression must either be null or a function"); } subClass.prototype = Object.create(superClass && superClass.prototype, { constructor: { value: subClass, writable: true, configurable: true } }); if (superClass) _setPrototypeOf(subClass, superClass); }
function _wrapNativeSuper(Class) { var _cache = typeof Map === "function" ? new Map() : undefined; _wrapNativeSuper = function _wrapNativeSuper(Class) { if (Class === null || !_isNativeFunction(Class)) return Class; if (typeof Class !== "function") { throw new TypeError("Super expression must either be null or a function"); } if (typeof _cache !== "undefined") { if (_cache.has(Class)) return _cache.get(Class); _cache.set(Class, Wrapper); } function Wrapper() { return _construct(Class, arguments, _getPrototypeOf(this).constructor); } Wrapper.prototype = Object.create(Class.prototype, { constructor: { value: Wrapper, enumerable: false, writable: true, configurable: true } }); return _setPrototypeOf(Wrapper, Class); }; return _wrapNativeSuper(Class); }
function isNativeReflectConstruct() { if (typeof Reflect === "undefined" || !Reflect.construct) return false; if (Reflect.construct.sham) return false; if (typeof Proxy === "function") return true; try { Date.prototype.toString.call(Reflect.construct(Date, [], function () {})); return true; } catch (e) { return false; } }
function _construct(Parent, args, Class) { if (isNativeReflectConstruct()) { _construct = Reflect.construct; } else { _construct = function _construct(Parent, args, Class) { var a = [null]; a.push.apply(a, args); var Constructor = Function.bind.apply(Parent, a); var instance = new Constructor(); if (Class) _setPrototypeOf(instance, Class.prototype); return instance; }; } return _construct.apply(null, arguments); }
function _isNativeFunction(fn) { return Function.toString.call(fn).indexOf("[native code]") !== -1; }
function _setPrototypeOf(o, p) { _setPrototypeOf = Object.setPrototypeOf || function _setPrototypeOf(o, p) { o.__proto__ = p; return o; }; return _setPrototypeOf(o, p); }
function _getPrototypeOf(o) { _getPrototypeOf = Object.setPrototypeOf ? Object.getPrototypeOf : function _getPrototypeOf(o) { return o.__proto__ || Object.getPrototypeOf(o); }; return _getPrototypeOf(o); }

// ACTUAL CODE STARTS HERE

var FileNotFoundError = function (_Error) {
  _inherits(FileNotFoundError, _Error);

  function FileNotFoundError(file, type) {
    var _this;

    _classCallCheck(this, FileNotFoundError);

    _this = _possibleConstructorReturn(this, _getPrototypeOf(FileNotFoundError).call(this, 'File not found: ' + file + ', type "' + type + '"'));
    _this.name = 'FileNotFoundError';
    _this.file = file;
    _this.type = type;
    return _this;
  }

  return FileNotFoundError;
}(_wrapNativeSuper(Error));

// DukTape error "unsupported"
new FileNotFoundError('filename', 'type');
```

Result of running this with lowjs-linux-x86_64-20181209:

```
Error: unsupported
    at [anon] (deps/duktape/src-low/duktape.c:62867) internal
    at _construct (/home/mha/neo/DEVICE/node_modules/one/lib/storage-base-common.js:10) strict tailcall
    at FileNotFoundError (/home/mha/neo/DEVICE/node_modules/one/lib/storage-base-common.js:23) strict construct
    at [anon] (/home/mha/neo/DEVICE/node_modules/one/lib/storage-base-common.js:33) strict preventsyield
```

lll000111, 2018-12-12:
Update: Babel 6 generated code works, but this Babel 7 code does not.

lll000111, 2018-12-12:
After replacing the Babel 7 class code with Babel 6 code I got past the above error (which still stands though, I think we really need a solution for Babel 7) I ran into a problem with the `Map` polyfill, which I think (without having investigated deeply) does not work well with the Babel 7 generated code using it.

The output starts with two lines from `console.log` statements I inserted just before the given location.

```
mha@mint-virtual-machine ~/neo/DEVICE $ ~/lowjs-linux-x86_64-20181212/bin/low index.js 
1 { next: [Function] }
1 undefined
TypeError: undefined not callable (property [Symbol wellknown '?Symbol.iterator?'] of [object Object])
    at [anon] (deps/duktape/src-low/duktape.c:64945) internal
    at informSubscribers (/home/mha/neo/DEVICE/node_modules/one/lib/version-map-updater.js:80) strict
    at _callee$ (/home/mha/neo/DEVICE/node_modules/one/lib/version-map-updater.js:162) strict
    at tryCatch (/home/mha/neo/node_modules/regenerator-runtime/runtime.js:62) strict
    at invoke (/home/mha/neo/node_modules/regenerator-runtime/runtime.js:296) strict tailcall
    at asyncGeneratorStep (/home/mha/neo/DEVICE/node_modules/@babel/runtime/helpers/asyncToGenerator.js:3)
    at _next (/home/mha/neo/DEVICE/node_modules/@babel/runtime/helpers/asyncToGenerator.js:25)
    at invokeCallback (lib:init:311) strict
    at publish (lib:init:381) strict
    at publishFulfillment (lib:init:386) strict
    at asyncFlush (lib:init:276) strict preventsyield
    [...]Error
```

The code at the location, including the `console.log` statements &mdash; line 80 is the `for` statements;

```js
  console.log(1, subscriptions.entries());
  console.log(1, subscriptions.entries()[Symbol.iterator]);
  try {
    for (var _iterator = subscriptions.entries()[Symbol.iterator](), _step; !(_iteratorNormalCompletion = (_step = _iterator.next()).done); _iteratorNormalCompletion = true) {
  ...
```

lll000111, 2018-12-12:
NOW RUNNING ON THE DEVICE (using `lowsync`)

I'm trying to use `lowsync` with un-babeled source code (only Flow type annotations and comments removed). The main problem here is that I get error messages that I cannot interpret because the line numbers are useless, I cannot see the code that is actually being run.

Right now I get

```
[1:28:50 PM] TypeError: cannot read property 0 of undefined
             at [anon] internal
             at [anon] (lib:init:489) strict preventsyield
             at forEach () native strict preventsyield
             at v (lib:init:489) strict
             at b (lib:init:488) strict construct
             at [anon] (/node_modules/one/lib/type-checks.js:148) strict
             at [anon] (/node_modules/one/lib/type-checks.js:149) strict preventsyield
             at require () native strict preventsyield
             at [anon] (/node_modules/one/lib/object-recipes.js:27) strict preventsyield
             at require () native strict preventsyield
             at [anon] (/node_modules/one/lib/instance.js:18) strict preventsyield
             [...]
```

neoniousTR, 2018-12-13:
just noticed there is more than the initial error.
will fix more tomorrow. goal is to get all of these bugs away + fs functions built till end of the coming weekend

neoniousTR, 2018-12-16:
Regarding the TypeError: undefined not callable (property [Symbol wellknown '?Symbol.iterator?'] of [object Object]) problem:

It seems like the code is calling [Symbol.iterator]() on a map iterator. So creating an iterator of an iterator.
This actually works in node:

new Map()[Symbol.iterator]()[Symbol.iterator]()[Symbol.iterator]()[Symbol.iterator]()[Symbol.iterator]()

is valid.

Would like to update low.js so this also works here. Right now only an iterator of a map can be created. But what is an iterator of an iterator of a map? Is it just a copy of the iterator, pointing to the same position the original iterator is also in? Cannot find anything about this online. Any idea?

Regarding the second problem with the console.log(10, 'Adding to Map:', e); :
This code is valid ES5.1, and runs with low.js on PC directly. Babel transform does not change it. And it runs through.

So weird that the behaviour on the board is supposedly different. Will try out tomorrow.

lll000111, 2018-12-16:
Babel bug?

neoniousTR, 2018-12-18:
I cannot reproduce your problem. Here, both with the console.log and without the output is:

[01:03:35] 11 [ [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ],
[01:03:35]   [ 's1', 's2' ] ]

... can you retry?

lll000111, 2019-01-16:
Update: It's on my TODO list, as soon as I can get the device working again...

closed 2019-03-20 by neoniousTR

Hypfer, 2020-05-16:
This seems to be still happening with the latest version

```
unhandledRejection Error: unsupported
    at [anon] (deps/duktape/src-low/duktape.c:41429) internal
    at get () native strict preventsyield
    at _get (/mnt/d/WebstormProjects/Valetudo_Github/dist/index.js:1) strict
    at parseStatus (/mnt/d/WebstormProjects/Valetudo_Github/dist/index.js:4050) strict
    at updateStatus (/mnt/d/WebstormProjects/Valetudo_Github/dist/index.js:2275) strict
```

The relevant code fragment in line 4050 looks like this:
`_get(_getPrototypeOf(RoborockS5.prototype),"parseStatus",this).call(this,res);`

commit 67126b22 (2018-12-13): fixed #35