neonious/lowjs #37: Documentation: timeline for in-development/upcomming features

domingosl, 2018-12-14:
I need some Node API that at the moment are not implemented or partially implemented (TLS and Crypto), to have a public timeline of developing milestones will be useful not only for me but for the community waiting for some API to start their projects/products.

neoniousTR, 2018-12-15:
There will not be a timeline, as there is currently no paid development. Everything will be done, but at a flexible schedule.

TLS and Crypto are both partly implemented, correct.
If you need something specific, it would be great to know what it is exactly...?

neoniousTR, 2018-12-18:
OK I understand.

Next on our list is upgrading DukTape to using a JIT compiler. Hopefully this will be done by end of the year. After that on the list is I2S, signal and touch modules.
On another team member's list is to upgrade the neonious one browser IDE so that TypeScript is compiled while saving, making neonious one the first microcontroller to support TypeScript out of the box.

Can you write a small script which runs with Node.JS just with the bits and pieces you need from low.js which are missing? This I could use to see if we can squeeze this into the timeline. Maybe it is not that much which has to be done.

domingosl, 2018-12-19:
I don't have a JS script ready yet, only on Arduino C, but I can already tell you what NodeJS API will be required. Essentially it is a TCP client over a secure channel (TLS), it connects to a server (wrote in NodeJS BTW) and then it exchanges messages using JSON, the JSON parsing is done using ArduinoJSON which I can simply replace with the JSON implementation in JS, the connection is handled using `WiFiClientSecure` that is replaceable with `tls.TLSSocket` in NodeJS and there is verification of the certificate fingerprint using `WiFiClientSecure::verify` which if I'm not wrong can be replaced by tlsSocket.getPeerCertificate in NodeJS.

neoniousTR, 2018-12-19:
So you are doing a custom TLS over TCP protocol, which is not HTTPs?
What do you need crypto then for?

In this case, it just seems we need to make sure tls is more fully 
implemented, correct? Because net already works 100%.

Thomas

closed 2019-03-20 by neoniousTR