neonious/lowjs #39: Set: TypeError: Invalid attempt to spread non-iterable instance

lll000111, 2018-12-23:
I'm using Babel 7, maybe there is an incompatibility with the polyfills provided by low.js?

```
TypeError: Invalid attempt to spread non-iterable instance
    at _nonIterableSpread (/home/mha/neo/DEVICE/node_modules/@babel/runtime/helpers/nonIterableSpread.js:2)
    at _toConsumableArray (/home/mha/neo/DEVICE/node_modules/@babel/runtime/helpers/toConsumableArray.js:8)
    at _send (/home/mha/neo/DEVICE/node_modules/one/lib/message-bus.js:68) strict tailcall
    at _callee2$ (/home/mha/neo/DEVICE/node_modules/one/lib/instance.js:73) strict
    at tryCatch (/home/mha/neo/node_modules/regenerator-runtime/runtime.js:62) strict
    at invoke (/home/mha/neo/node_modules/regenerator-runtime/runtime.js:288) strict tailcall
    at asyncGeneratorStep (/home/mha/neo/DEVICE/node_modules/@babel/runtime/helpers/asyncToGenerator.js:3)
    at _next (/home/mha/neo/DEVICE/node_modules/@babel/runtime/helpers/asyncToGenerator.js:25)
```

Code in `one/lib/message-bus.js` produced by Babel 7:

```js
  var typeHandler = handlers.get(type);

  if (typeHandler instanceof Set) {
    callbacks.push.apply(callbacks, (0, _toConsumableArray2.default)(typeHandler.values()));
  }
```
Let's follow what happens down in `_toConsumableArray2` and below in the Babel helpers. Code in `@babel/runtime/helpers/toConsumableArray.js:8`:

```js
var arrayWithoutHoles = require("./arrayWithoutHoles");

var iterableToArray = require("./iterableToArray");

var nonIterableSpread = require("./nonIterableSpread");

function _toConsumableArray(arr) {
  return arrayWithoutHoles(arr) || iterableToArray(arr) || nonIterableSpread();
}

module.exports = _toConsumableArray;
```

So that ends up failing in `iterableToArray`, because the final call to `nonIterableSpread()` has only a single active line of code, which is the `throw` instruction for the error.

Code in `@babel/runtime/helpers/iterableToArray.js` that should have worked but failed:

```js
function _iterableToArray(iter) {
  if (Symbol.iterator in Object(iter) || Object.prototype.toString.call(iter) === "[object Arguments]") return Array.from(iter);
}

module.exports = _iterableToArray;
```

lll000111, 2019-01-03:
NEW ATTEMPT: Same Babel and Babel config as lowsync uses.

Babel version taken from here: https://github.com/neonious/lowsync/blob/master/package.json#L28
Babel configuration taken from here: https://github.com/neonious/lowsync/blob/master/src/commands/commands/sync/synchronize/syncFile.ts#L69 (I added `"babel-plugin-transform-flow-strip-types": "^6.22.0"`)

Result:

```
mha@mint-virtual-machine ~/neo/DEVICE $ ~/lowjs-linux-x86_64-20190102/bin/low index.js 
START
Logger started. info logger
TypeError: undefined not callable (property [Symbol wellknown '?Symbol.iterator?'] of [object Object])
    at [anon] (deps/duktape/src-low/duktape.c:64945) internal
    at informSubscribers (/home/mha/neo/DEVICE/node_modules/one.core/lib/version-map-updater.js:314) strict
    at _callee$ (/home/mha/neo/DEVICE/node_modules/one.core/lib/version-map-updater.js:90) strict
    at tryCatch (lib:init:998) strict
    at invoke (lib:init:1195) strict tailcall
    at step (/home/mha/neo/DEVICE/node_modules/one.core/lib/version-map-updater.js:273) strict
    at [anon] (/home/mha/neo/DEVICE/node_modules/one.core/lib/version-map-updater.js:280) strict
    at invokeCallback (lib:init:310) strict
    at publish (lib:init:380) strict
    at publishFulfillment (lib:init:385) strict
    at asyncFlush (lib:init:275) strict preventsyield
    [...]Error
    at _callee2$ (/home/mha/neo/DEVICE/node_modules/one.core/lib/version-map-updater.js:372) strict
    at tryCatch (lib:init:998) strict
    at invoke (lib:init:1195) strict tailcall
    at step (/home/mha/neo/DEVICE/node_modules/one.core/lib/version-map-updater.js:273) strict
    at [anon] (/home/mha/neo/DEVICE/node_modules/one.core/lib/version-map-updater.js:282) strict
    at invokeCallback (lib:init:310) strict
    at publish (lib:init:380) strict
    at publishRejection (lib:init:389) strict
    at asyncFlush (lib:init:275) strict preventsyieldError
    at [anon] (/home/mha/neo/DEVICE/node_modules/one.core/lib/util/promise.js:297) strict
    at invokeCallback (lib:init:310) strict
    at publish (lib:init:380) strict
    at publishRejection (lib:init:389) strict
    at asyncFlush (lib:init:275) strict preventsyieldError
    at _callee2$ (/home/mha/neo/DEVICE/node_modules/one.core/lib/util/promise.js:111) strict
    at tryCatch (lib:init:998) strict
    at invoke (lib:init:1195) strict tailcall
    at step (/home/mha/neo/DEVICE/node_modules/one.core/lib/util/promise.js:249) strict
    at [anon] (/home/mha/neo/DEVICE/node_modules/one.core/lib/util/promise.js:258) strict
    at invokeCallback (lib:init:310) strict
    at publish (lib:init:380) strict
    at publishRejection (lib:init:389) strict
    at asyncFlush (lib:init:275) strict preventsyieldError
    at _callee$ (/home/mha/neo/DEVICE/node_modules/one.core/lib/storage-versioned-objects.js:127) strict
    at tryCatch (lib:init:998) strict
    at invoke (lib:init:1195) strict tailcall
    at step (/home/mha/neo/DEVICE/node_modules/one.core/lib/storage-versioned-objects.js:338) strict
    at [anon] (/home/mha/neo/DEVICE/node_modules/one.core/lib/storage-versioned-objects.js:347) strict
    at invokeCallback (lib:init:310) strict
    at publish (lib:init:380) strict
    at publishRejection (lib:init:389) strict
    at asyncFlush (lib:init:275) strict preventsyieldError
    at _callee3$ (/home/mha/neo/DEVICE/node_modules/one.core/lib/plan-storage-api.js:150) strict
    at tryCatch (lib:init:998) strict
    at invoke (lib:init:1195) strict tailcall
    at step (/home/mha/neo/DEVICE/node_modules/one.core/lib/plan-storage-api.js:31) strict
    at [anon] (/home/mha/neo/DEVICE/node_modules/one.core/lib/plan-storage-api.js:40) strict
    at invokeCallback (lib:init:310) strict
    at publish (lib:init:380) strict
    at publishRejection (lib:init:389) strict
    at asyncFlush (lib:init:275) strict preventsyieldError
    at _callee$ (/home/mha/neo/DEVICE/node_modules/one.core/lib/instance-updater.js:67) strict
    at tryCatch (lib:init:998) strict
    at invoke (lib:init:1195) strict tailcall
    at step (/home/mha/neo/DEVICE/node_modules/one.core/lib/instance-updater.js:235) strict
    at [anon] (/home/mha/neo/DEVICE/node_modules/one.core/lib/instance-updater.js:244) strict
[... 60 lines cut ...]
                fn = _step$value[0],
                hashes = _step$value[1];

            if (hashes === undefined || hashes.has(o.idHash)) {
                fn(o);
            }
        }
    } catch (err) {
        _didIteratorError = true;
        _iteratorError = err;
    } finally {
        try {
            if (!_iteratorNormalCompletion && _iterator.return) {
                _iterator.return();
            }
        } finally {
            if (_didIteratorError) {
                throw _iteratorError;
            }
        }
    }
}
```

So iterator issues again.

neoniousTR, 2019-01-07:
Is this urgent or are you having progress with your project even without this (maybe with workarounds?) This is on our road map but other things seem more important right now.

lll000111, 2019-01-07:
Well, we have quite *a lot* of `Set` and `Map` objects and iterations...

neoniousTR, 2019-01-07:
OK, I will fix things this week. Right now I do not know where to start however... Can you give a complete project, which I can fix? You know my email address.

lll000111, 2019-01-09:
No more error, the initial test using the Linux binary works! Has the device image been updated to include this patch yet?

lll000111, 2019-01-10:
May be network related.

After factory reset performed according to https://www.lowjs.org/documentation/neonious-one.html WiFi works, but only port 8443, no reaction on port 8000. As soon as I plug in the Ethernet I get the on/off cycle again, with a running ping showing "timeout" and responses in intervals.

neoniousTR, 2019-01-10:
Well, should not happen. Most probably related to the new JIT compiler which takes too much mem.
Will try out tonight, should be easy to fix. But good that the device is not bricked.

Sorry that you cannot continue directly.

lll000111, 2019-01-24:
@neoniousTR I just want to make sure my email from yesterday did not land in your spam folder...

> Hallo Thomas,
>
> Im Changelog steht seit dem Update, nachdem ich nicht mehr aufs Gerät zugreifen konnte, nichts neues drin. Ich frage daher lieber noch einmal nach, bevor ich den Update des gerade erhaltenen neuen boards starte...
>
> Beste Grüße

neoniousTR, 2019-05-08:
-- just a note that this bug is fixed and closed, as I am seeing many people coming here from Google --

mikeRChambers610, 2020-04-09:
This is still an issue I am having when trying to pull the data returned from a promise in react native.