neonious/lowjs #40: low.js binary enters busy loop when I do "require("core-js/fn/set");"

lll000111, 2018-12-23:
Command in the low.js REPL for easiest demonstration:

```
mha@mint-virtual-machine ~/neo/DEVICE $ ~/lowjs-linux-x86_64-20181223/bin/low 
> require("core-js/fn/set");

```

...and nothing. The process time keeps increasing steadily, but `strace -p` shows no activity. I cannot `kill`, I have to use `kill -9`.

core-js version is 2.5.7

lll000111, 2018-12-27:
Merry Christmas and a Happy New Year :-)

[image]

neoniousTR, 2018-12-28:
Merry christmas to you, too :-)

lll000111, 2018-12-28:
I just tried the latest Linux binary lowjs-linux-x86_64-20181228. The patch for #38 is in, but this one either is not, or it doesn't work, it still hangs. Since this patch is two hours older than the other one I would suspect this patch is in that binary?

neoniousTR, 2018-12-28:
My test run works, see below. This does not work on your system? It you are 100% sure, I will install a Linux virtual machine..

Last login: Fri Dec 28 12:52:36 on ttys001
Thomass-MacBook-Pro:~ thomas$ mkdir test
Thomass-MacBook-Pro:~ thomas$ cd test
Thomass-MacBook-Pro:test thomas$ npm init
This utility will walk you through creating a package.json file.
It only covers the most common items, and tries to guess sensible defaults.

See `npm help json` for definitive documentation on these fields
and exactly what they do.

Use `npm install <pkg>` afterwards to install a package and
save it as a dependency in the package.json file.

Press ^C at any time to quit.
package name: (test) 
version: (1.0.0) 
description: 
entry point: (index.js) 
test command: 
git repository: 
keywords: 
author: 
license: (ISC) 
About to write to /Users/thomas/Downloads/test/package.json:

{
  "name": "test",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "author": "",
  "license": "ISC"
}

Is this OK? (yes) yes
Thomass-MacBook-Pro:test thomas$ npm install core-js
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN test@1.0.0 No description
npm WARN test@1.0.0 No repository field.

+ core-js@2.6.1
added 1 package and audited 1 package in 2.047s
found 0 vulnerabilities

Thomass-MacBook-Pro:test thomas$ echo require\(\"core-js/fn/set\"\)\; > test.js
Thomass-MacBook-Pro:test thomas$ cat test.js 
require("core-js/fn/set");
Thomass-MacBook-Pro:test thomas$ node test.js 
Thomass-MacBook-Pro:test thomas$ ../lowjs-darwin-x86_64-20181228/bin/low test.js 
Thomass-MacBook-Pro:test thomas$

neoniousTR, 2018-12-28:
My escapes in the echo call were removed when posting here, but surely you understand the command.

lll000111, 2018-12-28:
1st console:

```
mha@mint-virtual-machine ~/neo $ ~/lowjs-linux-x86_64-20181228/bin/low 
> var fs = require('fs')
undefined
> require("core-js/fn/set");
Killed
```

2nd console:

```
mha@mint-virtual-machine ~/neo $ ps axf|grep lowjs-linux
 52920 pts/0    Rl+    0:07  |   \_ /home/mha/lowjs-linux-x86_64-20181228/bin/low

mha@mint-virtual-machine ~/neo $ grep version node_modules/core-js/package.json 
  "version": "2.5.7"
```

commit 6c2d9dd9 (2018-12-27): fixed #40
also using non-minified version of Set/Map for better debugging (production version is compiled anyhow)
commit dbf591f9 (2018-12-28): fixed #40 completly
only worked in test file before, not in console
also cleaned up vm: now using prototype instead of copying properties
closed 2018-12-28 by neoniousTR