neonious/lowjs #42: Plans to open source the ESP32 adaptation?

ernstnaezer, 2019-01-05:
Hi folks,

Like the title says - are there any plans to release the source code for adaptation of low to the ESP32 platform? And what would it take do so if not? I believe this would be a great start for new projects given the quality of the code I’ve seen thus far in the project. 

Thanks

neoniousTR, 2019-01-07:
Hi!

We can only spend a whole lot of time on low.js if we can build a business model out of it. And this is important because, to be realistic, people like to use open source, but only the largest projects really get meaningful contributions from other people outside the core dev group.

We think there is a target group for low.js on ESP32 of companies which use low.js to allow the end consumer to change things in JavaScript or TypeScript (maybe not at the lowest level, only modules) and thus make their product highly customisable. Companies can additionally license the neonious one IDE for this, allowing the end consumer to modify code right in the browser, with a great debugger and more.
All of this only works if they cannot just take the code, compile themselves and run.

We will be starting to look for sales people for this in this month. So, sorry, no, the plans go in the other direction.

But low.js for ESP32 is still free to use for hobby projects. What reason in particular are you interested in the source for the adaptation?

Thank you,

Thomas

ernstnaezer, 2019-01-07:
Hi Thomas,

thanks for your reply to my question. I understand your need to build a viable business in order to create and maintain a quality code base. There are already enough under-staffed and/or abandoned projects in the world. Open sourcing a complicated code base without maintainers and a vision it naive at best. I can also see that putting your full product out in the open creates the risk of companies, like you said, just taking the code, compile it themselves and run - leaving you in the dark. 

However, perhaps there is a middle ground that would satisfy both the open source community as well as companies looking for the features you mentioned? 

Take for instance JetBrains, the creators of IDEA and many other developer tools. They offer a Community Edition of their IDEs. This comes with limited functionality, but enough to support hobbyist and small scale developers. The same holds true for Datastax's databases - most are free and/or open but the Enterprise Edition adds comprehensive security and data encryption features, along with support and training and other advanced constructions that companies are willing to pay for. 

In my personal opinion and experience, having an open source version of a product is a strong selling point. The developers can give it a try, and when things seem to fall in place, the business decides to buy. This is especially true for companies that look for the advanced and/or unique features along with guaranteed support and maintenance agreements. 

I believe this hybrid model has lots of merits. So perhaps this is something for you to consider as well? Bring the IDE and features that allow a product to be highly customisable to market and offer a bare-bones edition to the community - the quality of your code base signals a valuable starting point for many. 

That being said, I of course cannot speak for your particular situation. It is your time, effort and code. And should you decide to move into the opposite direction I can only respect that. 

The particular reason for my request is twofold. I'm learning the ESP platform and looking at existing projects is of great personal benefit. Next to that, I'm looking to develop an application that relies on a fast and smooth UI (using a TFT screen). Achieving that performance requires finding the balance between parts written in C and parts written in JS. Having a project that I can compile allows for this flexibility. 

regards,
Ernst

neoniousTR, 2019-01-07:
Don't we have a hybrid model already with the non-ESP32 version online? The only part non-Open-Source is the part required to get low.js running well (running **well** is the difficult part) on ESP32.

We could add an official module system for the ESP32, where binary modules can be compiled and loaded? If yes, I would change this issue to that topic, however, to be honest, other things are far more important for now.

ernstnaezer, 2019-01-07:
Hi Thomas,

like I said, I cannot decide want makes the most sense for your situation. If there are no plans to open up (even a basic edition) of the ESP version, that is fine as well - I'll move along :) 

cheers,
Ernst

closed 2019-01-07 by ernstnaezer