neonious/lowjs #44: Issue with uart module and ESP32

anteph, 2019-01-25:
Hi,

I'm new to low.js and I'm trying to run a simple "Hello world" program. 
For that effect, I'm trying to import the 'uart' module to write some content to the serial port.

Nonetheless, when I import it, the program seems to stop working correctly. 

For example, the code below works well and turns on my board's LED:

```javascript
let gpio = require('gpio');

let state = 1;
gpio.pins[21].setType(gpio.OUTPUT).setValue(state);

setInterval(() => {}, 100);
```
If I add the import of the 'uart' module, then the LED no longer turns one, thus seeming to be causing some problem in the program.

```javascript
let uart = require('uart');
let gpio = require('gpio');

let state = 1;
gpio.pins[21].setType(gpio.OUTPUT).setValue(state);

setInterval(() => {}, 100);
```

I probably just missed some detail on the documentation, but if you could give me some help it would be awesome :)

Board: ESP32 TTGO T8 V1.1 (ESP32-WROVER module)

Best regards,
Nuno Santos

neoniousTR, 2019-01-27:
OK, this is an interesting problem. I'd be happy to find the issue very quickly...

Some questions:
(1) Is the board still responding to "lowsync status"? What does this say?
(2) What is the output of "lowsync monitor"?

neoniousTR, 2019-01-28:
Problem fixed in commit a77214cc2e9c981df14ff715f113d4e2820162d2 , see
https://github.com/neonious/lowjs/commit/a77214cc2e9c981df14ff715f113d4e2820162d2

Please retry!

closed 2019-01-28 by neoniousTR

anteph, 2019-01-28:
Sorry for the delay, just had the chance to get back to it today. 

I've just tested and it is working now, thank you very much :)

Best regards,
Nuno Santos