neonious/lowjs #45: How-to-use question (documentation issue)

lll000111, 2019-01-26:
I just replaced the entire existing demo code with our own stuff, which does not start a webserver. Now I can't reach the device (not a *hard* issue, factory reset works because this time the firmware works).

Do I have to have a webserver running from *my* code, even if I don't need it? If that is so, there seems to be some *minimum code* that always has to be present, which should be documented. I wish `lowsync` commands would still work (start, stop, status, sync, monitor) regardless of what my code does. My test ran some batch stuff, just create some files, and then it stopped (I think, I don't know, I can't access the device any more).

neoniousTR, 2019-01-27:
Do you have a copy of the code for me to try it out? Send per e-mail, if possible. Will keep it confidential.

lll000111, 2019-01-27:
Oh great

> Sorry, we were unable to deliver your message to the following address.

> <thomas@neonious.com>:
> 552: 5.7.0 This message was blocked because its content presents a potential
> 5.7.0 security issue. Please visit
> 5.7.0  https://support.google.com/mail/?p=BlockedMessage to review our
> 5.7.0 message content and attachment content guidelines. t7si2197939qvh.32 - gsmtp

lll000111, 2019-01-27:
Dear Mr. Geck, that is doch längst erledigt.

neoniousTR, 2019-01-28:
Thank you for the code. Problem found and fixed. Please install update.

neoniousTR, 2019-01-28:
Oh, forgot: Of course the IDE should always be available even if the code doesn't use a webserver. I am very sorry that low.js still is fragile. We are doing our best...

lll000111, 2019-01-28:
That's okay, I don't expect miracles :)

lll000111, 2019-01-29:
@neoniousTR  It does not work yet. Code still is the same that I sent.

I tried several times. The first few times it only seems to have been a "soft crash" &mdash; after a few seconds the device was reachable again. Looking at the filesystem with the IDE showed that no files had been created in the /DATA/ directory so the crash must have come early.

After the fifth or so time now of doing the same, starting the code at `/index.js` with `lowsync start`, the device now is unreachable for good without a power off/on cycle.

neoniousTR, 2019-01-29:
I uploaded the project with the IDE (which transpiles it) and tried it out. There was a problem with the stack compression (this is really low level code), which I fixed. It worked stable then.

I guess I should upload the project with lowsync without transpiling, as that is what you are doing? Will do that tonight, and will fix any problems then.

neoniousTR, 2019-01-29:
Yes, but the IDE uses TypeScript and not Babel, as you know (I know, I know, should be changed..)

So I will try with the exact same settings as you have tonight.

lll000111, 2019-01-29:
Here is my build.js file: https://gist.github.com/lll000111/1b5487dc6bd07bd8b798083e4149bfa2

lll000111, 2019-01-29:
I just updated one.core and this time, when running the same index.js through the Linux low.js binary, I find that there is no `Array.prototype.includes`. I did not get that error previously, but I *do* use `includes` a few times throughout the one.core lib. Do I have to polyfill that method myself?

lll000111, 2019-01-29:
I'll send an updated archive later, now with a simple polyfill for `includes` and the latest code I'm testing with.

neoniousTR, 2019-01-30:
Can reproduce the bug and can trace it to the stack compression.
The C code has side effects which only show in the release version. Will rewrite it in ASM ASAP (today or tomorrow).
Thomas

neoniousTR, 2019-01-30:
Fixed! At least so I think.
- There was a problem with the stack compression, as I wrote
- Also fs.access did not work correctly on ESP32
Please update.

You will however notice two things which you might not like:
- Your fs.open does not work b/c your string is too long. We currently only support files with absolute paths <= 127 chars
- It is slow loading all your modules. Sorry for that. Optimizing is on our list..

lll000111, 2019-01-31:
> It is slow loading all your modules. Sorry for that. Optimizing is on our list..

Does that mean "webpacking" it all into a single file would be better?

neoniousTR, 2019-01-31:
Will definitly help massively, the file system is slow. But how much I cannot say yet. Next on the list: Get native npm with GUI IDE and lowsync optimisations working.

lll000111, 2019-02-02:
I'll just continue posting here.

The latest crash:

```
[12:29:43 PM] Error: memory full
              at [anon] internal
```

Code is the same as before, but now I put it into a single bundle file created with *rollup*.

Runs fine using the Linux low.js binary.

I sent the bundle file cloud link by email, as always.

I powered on/off, same result.

lll000111, 2019-02-02:
PS: The IDE's resource monitor shows plenty of memory still available.

neoniousTR, 2019-02-02:
I am sorry, but it works perfectly here... Can you pinpoint the location of the problem more?

lll000111, 2019-02-02:
I copy-pasted the error message. All I do is run the file.

lll000111, 2019-02-02:
By the way, that file is the only file in the file system (that is accessible to me via "lowsync sync" and/or IDE File Explorer), auto-start is turned off, "lowsync monitor" runs, I start the file - after a power off/on cycle, and that is the message that I get, nothing else, shortly after "lowsync start".

neoniousTR, 2019-02-09:
OK, seems to be a problem by design: memory fragmentation.

I really like the way the old Mac OS worked... With relocatable handles which allow memory to be moved when not enough space is available. But, to implement this, all software has to be changed to make use of this.

So, the way it is right now, allocating a large buffer (1 MB is the buffer it tries to allocate here) sometimes works, sometime does not. This means that large JS files cannot be loaded, as for this it loads the file in memory completly.

So, we need to make it faster the way it is. One bottleneck definitly is the file system: The new update (will be online hopefully in the next 2 days) has the built in npm package manager. If installed from here, the whole module is loaded as one file (something like a ZIP). The modules load 3x as fast from here than if we have the files seperatly.

The file system we use is ESPIFFS. It seems not to be written with "larger" Flash in mind. I guess we will have to write a better FS ourselves some day.

lll000111, 2019-02-09:
Well, if I don't have to build a single file than that's all fine and not a priority I think, since few people will run into the issue in that case.

closed 2019-03-20 by neoniousTR