neonious/lowjs #47: Load node_modules folder from SD card

bart, 2019-01-28:
Hey guys,

I'm currently experimenting a lot with low.js. I want to use some node modules which have a size of 9 MB total. For sure that amount of data must be stored on sd card. It looks like you haven't yet implemented filesystem symlinking so it seems to be impossible to load node modules from sd card.

This is my current code:

```
const sdcard = require('sdcard');
const fs = require('fs');

sdcard.mount(() => {
    fs.symlink('./sd/node_modules', './node_modules', () => {
        console.log('Node modules folder symlinked successfully');

        // When it's implemented I can load modules here with require()
    });
});
```

Do you have any workaround for it? Btw. I must do the sdcard.mount() thing. Otherwise /sd is not mounted. Maybe due to a timing issue which starts index.js before SD card has been mounted.

neoniousTR, 2019-01-29:
Interesting that you have to call sdcard.mount... But apart from that, it works for you? What boad are you using and what settings? Will check this.

Regarding symlinks: Not a high prio here.

Workaround:
var mounter = require(./sdcard/mounter.js) and call exports.themodule == require('themodule') from there? It should read themodule from ./sdcard/node_modules then and you can access it in your main module via mounter.themodule.

Would that work?

bart, 2019-01-29:
Yeah, I tried it without the `sdcard.mount()` method and checked it with a simple file list operation which did not work. Just after adding the `sdcard.mount()`method it was able to list files. I'm using the TTGO-T8-ESP32 V1.7 (https://github.com/LilyGO/TTGO-T8-ESP32). I'll try the workaround later today or tomorrow. Thanks a lot!

toanphambk, 2020-10-02:
i wrote some js file in the sdcard and i try to import them to index.js file but it it get this error all the time

`SyntaxError: empty expression not allowed (line 4)
    at [anon] internal
    at [anon] internal
    at require () native strict preventsyield
    at [anon] (/index.js:11) strict preventsyield`

this is my index.js file

`const sdCard = require("sdcard");

sdCard.mount(() => {
  console.log("sdcard mounted");
  const { ioTest } = require("/sdcard/app.js");
  ioTest();
});

the  app.js  file in /sdcard/ (which is the folder i mount sd card to)

`const gpio = require("gpio");

module.exports = {
  ioTest: () => {
    gpio.pins[13].setType(gpio.OUTPUT);
    let val = 0,
      dir = 0;
  
    setInterval(() => {
      val += dir ? 0.03 : -0.03;
      if (val < 0) {
        val = 0;
        dir = 1;
      }
      if (val > 1) {
        val = 1;
        dir = 0;
      }
      gpio.pins[13].setValue(val);
    }, 30);
  },
};
`
`please help!!!

toanphambk, 2020-10-02:
i wrote some js file in the sdcard and i try to import them to index.js file but it it get this error all the time

`SyntaxError: empty expression not allowed (line 4) at [anon] internal at [anon] internal at require () native strict preventsyield at [anon] (/index.js:11) strict preventsyield`

this is my index.js file

```
const sdCard = require("sdcard");
sdCard.mount(() => {
console.log("sdcard mounted");
const { ioTest } = require("/sdcard/app.js");
ioTest();
});
```

the app.js file in /sdcard/ (which is the folder i mount sd card to)
```

const gpio = require("gpio");

module.exports = {
ioTest: () => {
gpio.pins[13].setType(gpio.OUTPUT);
let val = 0,
dir = 0;

setInterval(() => {
  val += dir ? 0.03 : -0.03;
  if (val < 0) {
    val = 0;
    dir = 1;
  }
  if (val > 1) {
    val = 1;
    dir = 0;
  }
  gpio.pins[13].setValue(val);
}, 30);
},
};
```

toanphambk, 2020-10-02:
i wrote some js file in the sdcard and i try to import them to index.js file but it it get this error all the time

```
SyntaxError: empty expression not allowed (line 4)
    at [anon] internal
    at [anon] internal
    at require () native strict preventsyield
    at [anon] (/index.js:11) strict preventsyield
Fatal error, restarting program...
```

this is my index.js file

const sdCard = require("sdcard");

```
sdCard.mount(() => {
  console.log("sdcard mounted");
  const { ioTest } = require("/sdcard/app.js");
  ioTest();
});
```

the app.js file in /sdcard/ (which is the folder i mount sd card to)

```
const gpio = require("gpio");
module.exports = {
  ioTest: () => {
    gpio.pins[13].setType(gpio.OUTPUT);
    let val = 0,
      dir = 0;
  
    setInterval(() => {
      val += dir ? 0.03 : -0.03;
      if (val < 0) {
        val = 0;
        dir = 1;
      }
      if (val > 1) {
        val = 1;
        dir = 0;
      }
  
      gpio.pins[13].setValue(val);
    }, 30);
  },
};
```

ThomasRogg, 2020-10-05:
Hi!

I expect you put the second file on the SD Card directly. This way the code is not getting transcoded (lowsync and the neonious IDE do it on the fly), so you have to either transcode yourself in a build step or use ES5.1 only.

Regards

Thomas

toanphambk, 2020-10-09:
Thank you Thomas, I have try to use babel a few day ago and it works like a charm. Thank you very much!

> Hi!
> 
> I expect you put the second file on the SD Card directly. This way the code is not getting transcoded (lowsync and the neonious IDE do it on the fly), so you have to either transcode yourself in a build step or use ES5.1 only.
> 
> Regards
> 
> Thomas

closed 2021-02-21 by ThomasRogg