neonious/lowjs #48: I2C usage

bart, 2019-01-28:
Hey guys, unfortunately I wasn't able to figure out how the I2C module with lowjs works. Normally these i2c packages have methods like `writeByte` and `readByte` but here it seems to be `transfer` only. May you give an example for it? That would be very kind of you.

Thanks you very much!

neoniousTR, 2019-01-29:
Any I2C module in mind? Otherwise I will add on the list to create an example for the ADXL345 accelerator this week.

bart, 2019-01-29:
Hey @neoniousTR Thanks a lot for your response. I want to use the PCA9685. Unfortunately it currently looks like I must switch to MicroPython because hardware related Javascript seems to be a bit difficult :(

neoniousTR, 2019-01-30:
Have example on my list. But till then, what's the exact difficulty you have?

With transfer you give a Buffer which is written, and a Buffer which is then filled with read data. This is how I2C works: A read phase and a write phase.

neoniousTR, 2019-02-06:
Said it the wrong way around: Normally you have a write phase, and then a read phase.

OK here is and example:
Writes 1 byte with the value 23 to the I2C address 0x53 and reads 1 byte afterwards.

```
let i2cMod = require('i2c');
let i2c = new i2cMod.I2C({
    pinSCL: 20,
    pinSDA: 21
});

let dataWrite = Buffer.alloc(1 /* Number of bytes to write */);

// Set byte 0 to 23
dataWrite.writeUInt8(23, 0);

i2c.transfer(0x53, dataWrite, 1 /* Number of bytes to read */, (err, dataRead) => {
	if(err) {
		console.log("error: ", err);
		return;
	}

	console.log("byte read: ", data.readUInt8(0));
    }
});
```

closed 2019-02-06 by neoniousTR

bart, 2019-02-07:
Awesome, thanks a lot @neoniousTR !