neonious/lowjs #50: IDE: "File Explorer" does not show all files

lll000111, 2019-01-31:
I ran the mini app test and it works now, but if I want to check if the files it was supposed to create were in fact created I have to do a `lowsync sync` to sync them back from the device into the PC sync directory. Looking at the file system with the IDE's "File Explorer" show the directories under "/DATA/...", but the directories are shown as empty. They contain 64 hex-character named files without extension, e.g. "44af76dd9c44e044d2346c4eac40e4aaff08f2f163573181939e8521ea29f8d7". Those are (UTF-8) text files stored under the SHA-256 of their contents.

Does the File Explorer filter the list of files that it displays?

<br>

PS: Latest code sent as link to archive, just FYI, I guess you can answer this one without having to execute anything. I changed the code to limit the path length as advised.

neoniousTR, 2019-01-31:
So the files ARE created because you get them with lowsync, but you do not see them in the IDE?

lll000111, 2019-01-31:
Yes, it seems that it works.

There is nothing in /DATA/ on the PC, as shown in the first line below, then I run the code on the device, the directories appear in the IDE but without the arrow that would let me open them, then I sync and teh PC's /DATA/ directory is filled with the files.

```
mha@mint-virtual-machine ~/neo/DEVICE $ l DATA/
total 0
mha@mint-virtual-machine ~/neo/DEVICE $ lowsync sync
Fetching file system listings...
Synchronization complete |██████████████████████████████| 100%
MC => PC: +Folder DATA/objects
MC => PC: +File DATA/objects/44af76dd9c44e044d2346c4eac40e4aaff08f2f163573181939e8521ea29f8d7
MC => PC: +File DATA/objects/4705654ad015a0a2bed1f7badf84c3d2b9acb1e2e8fc412c8a5748c7183d96da
MC => PC: +File DATA/objects/8c5d4358cf8cc7a6b540de735b2931112468c4635b13b2c67abec800e688f81e
MC => PC: +File DATA/objects/a7aea63e320ab9abadb00c77d4bec1609be17a14015be890a0255569f6e43709
MC => PC: +Folder DATA/reverse-maps
MC => PC: +File DATA/reverse-maps/1d5a69cf5835a71166adfe702bdc52160492bb65d3acce6d782040abec21d47f.I.to.PlanEnd
MC => PC: +File DATA/reverse-maps/9c79ad64280eaaf20c6bee71ef65888d28a8cfae65cbb2317be3b76b71b1292a.I.to.Instance
MC => PC: +File DATA/reverse-maps/9c79ad64280eaaf20c6bee71ef65888d28a8cfae65cbb2317be3b76b71b1292a.I.to.PlanEnd
MC => PC: +Folder DATA/tmp
MC => PC: +Folder DATA/version-maps
MC => PC: +File DATA/version-maps/1d5a69cf5835a71166adfe702bdc52160492bb65d3acce6d782040abec21d47f
MC => PC: +File DATA/version-maps/9c79ad64280eaaf20c6bee71ef65888d28a8cfae65cbb2317be3b76b71b1292a
? Would you like to show the output of the microcontroller? (Use the --monitor command line option to enable or disable automatic showing of the output after sync.) No
mha@mint-virtual-machine ~/neo/DEVICE $ l DATA/
total 16
drwxr-xr-x 2 mha mha 4096 Jan 31 13:20 objects
drwxr-xr-x 2 mha mha 4096 Jan 31 13:20 reverse-maps
drwxr-xr-x 2 mha mha 4096 Jan 31 13:20 tmp
drwxr-xr-x 2 mha mha 4096 Jan 31 13:20 version-maps
mha@mint-virtual-machine ~/neo/DEVICE $ l DATA/*
DATA/objects:
total 16
-rw-r--r-- 1 mha mha 1401 Jan 31 13:20 44af76dd9c44e044d2346c4eac40e4aaff08f2f163573181939e8521ea29f8d7
-rw-r--r-- 1 mha mha  270 Jan 31 13:20 4705654ad015a0a2bed1f7badf84c3d2b9acb1e2e8fc412c8a5748c7183d96da
-rw-r--r-- 1 mha mha   98 Jan 31 13:20 8c5d4358cf8cc7a6b540de735b2931112468c4635b13b2c67abec800e688f81e
-rw-r--r-- 1 mha mha  194 Jan 31 13:20 a7aea63e320ab9abadb00c77d4bec1609be17a14015be890a0255569f6e43709

DATA/reverse-maps:
total 12
-rw-r--r-- 1 mha mha 131 Jan 31 13:20 1d5a69cf5835a71166adfe702bdc52160492bb65d3acce6d782040abec21d47f.I.to.PlanEnd
-rw-r--r-- 1 mha mha 195 Jan 31 13:20 9c79ad64280eaaf20c6bee71ef65888d28a8cfae65cbb2317be3b76b71b1292a.I.to.Instance
-rw-r--r-- 1 mha mha 131 Jan 31 13:20 9c79ad64280eaaf20c6bee71ef65888d28a8cfae65cbb2317be3b76b71b1292a.I.to.PlanEnd

DATA/tmp:
total 0

DATA/version-maps:
total 8
-rw-r--r-- 1 mha mha 147 Jan 31 13:20 1d5a69cf5835a71166adfe702bdc52160492bb65d3acce6d782040abec21d47f
-rw-r--r-- 1 mha mha 147 Jan 31 13:20 9c79ad64280eaaf20c6bee71ef65888d28a8cfae65cbb2317be3b76b71b1292a

```

neoniousTR, 2019-02-18:
I have to mark this an Enchancement and not a Bug.

Reason for this is that mkdir updates the neonious IDE as a side effect, but actually this was not intentional..

Anyways, will be fixed by doing this for file opening, too. Next on the list however is to support https clients.

neoniousTR, 2019-05-23:
Will be fixed with new faster file system we are currently implementing.

neoniousTR, 2020-05-15:
Finally, this is fixed with the new file system in the current version 1.5.0!

closed 2020-05-15 by neoniousTR