neonious/lowjs #53: KE32-WROVER can't connect to the the default gateway on the wifi.

DeaSTL, 2019-03-08:
[image]

neoniousTR, 2019-03-10:
Hello! I am not sure what you are trying to say. It looks like your local web browser cannot connect to 192.168.0.1. But what is 192.168.0.1 supposed to be? I need some more information. Thank you!

DeaSTL, 2019-03-10:
@neoniousTR  I've flashed the firmware to my esp32 Wrover and then connected to the wifi provided by the esp32-wrover and once i've entered the ip provided by the dialog in lowsync, it instantly disconnects.

neoniousTR, 2019-03-11:
I am sorry, but I do not understand your description. What disconnects? The Wifi? I cannot reproduce this. And what does that have to do with the screenshot? I cannot see that anything disconnects in the screenshot?

closed 2019-03-20 by neoniousTR