neonious/lowjs #55: spi module not found from TypeScript

Nervengift, 2019-03-18:
We just got a few neonious one for a hackathon and I wanted to try the TypeScript support in the IDE, but I'm failing to import the ESP32-specific modules like 'spi'. Standard modules like 'http' seem to work fine

```
import * as http from 'http';
http.createServer(function(req, res){res.end();});

import * as spi from 'spi';

let s = new spi.SPI({pinSCLK:7, pinMOSI:6});
s.transfer(Buffer.from([0x00, 0x00, 0x00, 0x00]));
```

Fails with `Cannot find module 'spi'. (4, 22)`.

The equivalent JavaScript (which is basically the same as the transpiled code) works as expected.

Am I missing something? The documentation is not really mentioning TypeScript at all...

neoniousTR, 2019-03-20:
How does the transpiled code look like? After saving you can do a right click on the file and select "Show transpiled code"
Is it only an error which you can see under Problems, oder is it really a failure when trying to run the program?

Nervengift, 2019-03-21:
```js
Object.defineProperty(exports, "__esModule", { value: true });
var http = require("http");
http.createServer(function (req, res) { res.end(); });
var spi = require("spi");
var s = new spi.SPI({ pinSCLK: 7, pinMOSI: 6 });
s.transfer(Buffer.from([0x00, 0x00, 0x00, 0x00]));
```

It won't start and gives an error dialog:
"An fatal error has occured in the program:
SyntaxError: parse error (line 1)
Program cannot continue."

Copying the generated code to a .js file and running it works as intended

neoniousTR, 2019-07-03:
We have to decided to remove TypeScript from the official support list. Just not enough people who request it, and so we need to concentrate on other things. See https://www.neonious.com/documentation/changelog on what we are doing.

This changes however as soon as someone is willing to sponsor TypeScript development. Till then, I am closing this issue.

closed 2019-07-03 by neoniousTR