neonious/lowjs #59: Now with TypeScript: but lots and lots of errors

lll000111, 2019-05-01:
Now we actually switched to TypeScript, so I thought since your IDE is focused on it we should have smoother sailing. I was wrong :(

I prepared the TS files and then uploaded them into /src on the Neo One, to continue with the IDE from there. Structure: index.ts in /src/ and a directory /src/one.core (plus, below, ./sytem/ and ./util/) with a lot of module files, all .ts.

After uploading I started the IDE and it started to compile the .ts files. Result: Lots and lots of errors for almost everything that I thought would be normal. Even basic TS types like Promise, Map, Set are unknown. The only imports that work are from index.ts to two one.core/ files, all relative imports inside that directory fail.

Output: https://2c2ba923-f87a-4112-8b33-4a621bcc8260.htmlpasta.com/

lll000111, 2019-05-01:
By the way, before uploading the .ts files I tried uploading the .js files created by my development environment's TypeScript transpiler.

The result: After the sync was done, 100% CPU for a minute, then "connection was lost" (both lowsync monitor and IDE), after a reconnect (automatic for lowsync monitor, manual restart of IDE) I saw - nothing. The code didn't produce the files it was supposed to.

The same very simple "only start the app and let it create its default files" had already worked a while ago, when I had issues with heap and cache sizes). So this is a step backwards.

lll000111, 2019-05-01:
PS: I compile with `tsc` for

```
"importHelpers": true,
"esModuleInterop": true,
"module": "CommonJS",
"moduleResolution": "node",
"target" "es5", 
"lib": ["esnext", "dom"],
```

and all I need is the (small) `tslib.js` imported by some files.

neoniousTR, 2019-07-03:
We have to decided to remove TypeScript from the official support list. Just not enough people who request it, and so we need to concentrate on other things. See https://www.neonious.com/documentation/changelog on what we are doing.

This changes however as soon as someone is willing to sponsor TypeScript development. Till then, I am closing this issue.

closed 2019-07-03 by neoniousTR

lll000111, 2019-07-04:
That's a good thing, I argued against it from the start if you remember :-)