neonious/lowjs #60: When is https client going to be supported?

SubaruWRX55, 2019-05-22:
We are working on a BLE bridge based on ESP32 and very interested in trying out low.js. One of the requirement is to communicate with our server securely via HTTPS. We use axios npm library, to be specific. 

What is the timeline for getting https running in lowjs?

neoniousTR, 2019-05-23:
This is not much work, but it is not on our priority list. You will see in https://www.neonious.com/documentation/changelog that we are moving fast right now, but the goal for the next months is 100% stability (right now it is solid, but not industry-standard-rock-solid) and speed.

Are you willing to donate/pay a minimum for this? We could change priority then. If interested, you can write to info@neonious.com

neoniousTR, 2019-06-19:
low.js now supports HTTPS clients and axios! @SubaruWRX55 I will send you an email about this. For others, see our newsletter for more details: https://www.neonious-basics.com/index.php/2019/06/19/newsletter-june-19th-2019/

closed 2019-06-19 by neoniousTR