neonious/lowjs #62: An error has occured: esptool exited with exit code 2. Exiting.

TAMHAN, 2019-06-12:
Hello Folks,
sorry to make a nuissance of myself, but I found an issue burning the ESP32 firmware to my WROVER module. On Ubuntu 18, I get the following error:
```
tamhan@TAMHAN18:~$ sudo lowsync flash /dev/ttyUSB0 --init
*** Step 1/3: Probing ESP32 microcontroller
    now checking if it is an ESP32-WROVER... (takes a while)
An error has occured: esptool exited with exit code 2. Exiting.

```

Actual communication with the GeekWorm module, however, works:
```
> tamhan@TAMHAN18:~$ esptool.py read_mac
> esptool.py v2.6
> Found 2 serial ports
> Serial port /dev/ttyUSB0
> Connecting........__
> Detecting chip type... ESP32
> Chip is ESP32D0WDQ6 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> MAC: 80:7d:3a:df:c2:24
> Uploading stub...
> Running stub...
> Stub running...
> MAC: 80:7d:3a:df:c2:24
> Hard resetting via RTS pin.
```

This is a bit frustrating. I already tried updating all the components and even installed the esptool twice. The error, however, remains constantly the same.

Strangely, the board does blink its LEDs a few times. The process also takes quite long:
```
tamhan@TAMHAN18:~$ time sudo lowsync flash /dev/ttyUSB0 --init
*** Step 1/3: Probing ESP32 microcontroller
    now checking if it is an ESP32-WROVER... (takes a while)
An error has occured: esptool exited with exit code 2. Exiting.

real	2m8,051s
user	0m0,516s
sys	0m0,080s

```

I have already tried the usual suspects. The cable is of good quality, and has been tested with other boards like an Azure Sphere.  Oddly, when connecting an ESP32 WROOM, the test fails much faster:

```
> tamhan@TAMHAN18:~$ time sudo lowsync flash /dev/ttyUSB0 --init
> *** Step 1/3: Probing ESP32 microcontroller
>     now checking if it is an ESP32-WROVER... (takes a while)
> An error has occured: ESP32 is not an ESP32-WROVER or at least does not have required 4 MB PSRAM!
> Please check: https://www.lowjs.org/supported-hardware.html
> 
> real	0m16,882s
> user	0m0,733s
> sys	0m0,110s
```

Tam

neoniousTR, 2019-06-13:
Hi!
Can you point me to the exact module you are using? A link to a page to buy or anything?
Thank you!
Thomas

TAMHAN, 2019-06-13:
Hello,
this is the board:
https://www.aliexpress.com/item/Geekworm-ESP32-WROVER-Development-Board-with-8MB-PSRAM-WiFi-Bluetooth-Low-Power-Consumption-Dual-Cores-ESP32/32902145720.html?spm=a2g0s.9042311.0.0.20524c4dvMtvsJ

If you want to, I can send you the board to play with it. I am using LowJS mainly to write a story in the PHPMagazin. 

Tam

neoniousTR, 2019-06-13:
Hello Tam,
well, if you are writing a story, I am eager to get this working so you can take a look.
The board seems basic. Maybe the problem is the 8 MB of RAM, but I do not see any other difference.
I think it could be easier to figure out if you help me with a few outputs...
It seems to flash the small program which checks whether the board is a WROVER. This program does not function, however, or so it seems. After it no longer responds, can you check what the program outputs with any terminal program? Just connect to the Serial Port and see what it outputs. It should either output NOT_WROVER or IS_WROVER in a loop, if it does not and outputs something else, this would help me.
If this does not help me, I think I have a 8 MB RAM WROVER board at home somewhere, I would try it out tonight.
Thomas

neoniousTR, 2019-06-13:
Wait a second.. esptool is exiting... Hmm.. maybe it did not flash after all. Checking at home.

TAMHAN, 2019-06-13:
Hello,
I tried to connect with a terminal, and only got garbage output. I tried various baud rates, always the same.

But: I am not a GitHub native. It is easier to communicate with me by email - tamhan@tamoggemon.com. 

Tam

neoniousTR, 2019-06-13:
> email - tamhan@tamoggemon.com <mailto:tamhan@tamoggemon.com>.

Hi,

flashing works perfectly here.

Can you try a reduced baud rate? Add a "-b 115200" to the command, 
please. Then, give it a minute.

Thomas

neonious GmbH

Münsterstr. 246 / 40470 Düsseldorf / Germany

T   +49 211 9241 8187
F   +49 211 9241 8172

www.neonious.com <http://www.neonious.com>

Managing Director & CEO: Thomas Rogg
Commercial Register: Amtsgericht Düsseldorf, HRB 83086

weixiongmei, 2019-06-15:
Hi, I'm having the same issue too, the board i have it's using the CH340 USB-UART chip, would that be a problem?

neoniousTR, 2019-06-15:
That can very much be the thing, the board of TAMHAN is also with the CH340 chip. Its being shipped to us, so I will be able to try it out in a few days, I will write then.

weixiongmei, 2019-06-16:
Hi @neoniousTR , i just sent you an e-mail, can you let me know you're interesting to fix it for me or not? Thanks~~

TAMHAN, 2019-06-17:
Hello,
I already sent him a board to evaluate.

neoniousTR, 2019-06-22:
OK, I found the problem.

lowsync uses esptool with the parameter -b 921600 as a standard. Adding other bitrates instead did not work either. But without any bitrate it works. Seems like the CH340 does not support as high bitrates as the other chips (FT232 and CP2102).

So, the newest version of lowsync (1.0.23) now works with this board. And if you want the flashing as fast as it was before, you can still add -b 921600 to the command line. But usually flashing is only done once anyways.

@TAMHAN : I am sending your board back to the address you sent to me by email now.

Thomas

closed 2019-06-22 by neoniousTR

Manifoltwarper, 2019-12-28:
Dear neoniousTR, I havent been able to understand how exactly you solved this could you please explain further?

waninlezu, 2020-02-22:
@neoniousTR same issue here:

```
$ lowsync flash --port=/dev/cu.SLAB_USBtoUART --init

*** Step 1/3: Probing ESP32 microcontroller
    now checking if it is an ESP32-WROVER... (takes a while)
Error: ESP32 is not an ESP32-WROVER or at least does not have required 4 MB PSRAM!
Please check: https://www.lowjs.org/supported-hardware.html
```

Chip info:

```
$ esptool.py   --port=/dev/cu.SLAB_USBtoUART flash_id
esptool.py v3.0-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 84:0d:8e:e3:aa:fc
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

> It seems to flash the small program which checks whether the board is a WROVER. This program does not function, however, or so it seems. After it no longer responds, can you check what the program [...]

the small program does not function and output as:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5212
load:0x40078000,len:9036
ho 0 tail 12 room 4
load:0x40080400,len:6572
entry 0x400806f8
^Es0Tp!jhq      cv!bFtfQv!r0GG`<!KanAJaPra{GjB!rXbp4uegb11b1bb
```