neonious/lowjs #63: Investigate QuickJS

stevefan1999-personal, 2019-07-12:
[Fabrice Bellard](https://bellard.org/), the creator of QEMU, FFMPEG and TinyCC, has released [QuickJS](https://bellard.org/quickjs/) which claims almost 100% ES2019 compliance. Maybe we could try and port the CommonJS/Node part of lowjs to see if it also works to make a better Node alternative?

neoniousTR, 2019-07-12:
Hi! First, I noticed you also did the CMake pull request which is on hold. Thank you for still taking a look at this project. 

QuickJS looks interesting, however is a project which I cannot do in our free time and I guess it will be hard to get people to pay for the port.
So if you or anybody else wants to start this, go ahead! But I am sorry that I have to pass for now.

BTW With such a large contribution such as an additional engine I guess I would take my time to get used to CMake, too..

neoniousTR, 2019-10-29:
Took a look at QuickJS. Some crucial things are still missing, especially the debugging interface. But monitoring the situation.

closed 2020-05-15 by neoniousTR