neonious/lowjs #66: Not able to sync

BergenSoft, 2019-09-13:
The `lowsync flash COM3 --init` command was successful, I got the WIFI SSID with password and could connect to it. The default IP of the board is shown as 192.168.0.1

When I open the Browser, I could see the page with the content:
"low.js / lowsync endpoint
This is the HTTP(S) server which handles lowsync requests."

So 

It works for both https://192.168.0.1:8443 and http://192.168.0.1:8000

But when I try to sync the local files `lowsync` I got the following message:
The device cannot be reached with the provided protocol, IP and port (192.168.0.1 8443 ssl: true).
The device cannot be reached with the provided protocol, IP and port (192.168.0.1 8000 ssl: false).

Depends if I test with ssl or not.

I could also change the wifi settings like described here: https://www.lowjs.org/documentation/flash-space.html

This is also working and I could connect with the browser to the IP in my network.

So the flashing and everythings looks that it works perfectly.

I'm using a **ESP32-WROVER-B** Board and **lowsync v.1.1.0**.
Could that cause the problems?

Thanks a lot.

neoniousTR, 2019-09-13:
I know that lowsync under Windows show "cannot be reached" errors, but it nonetheless works on my test system. See https://github.com/neonious/lowsync/issues/7 for more info.

Do I understand it correctly that it is the same on your system, that flashing works, but you are wondering about the errors? Or does the sync not work at all, but everything else (set settings and more) work?

Thank you
Thomas

BergenSoft, 2019-09-14:
The flashing works without error. But the sync or monitor command gives me the error that the microcontroller cannot be found. I will set up a virtual linux and test it from there.

BergenSoft, 2019-09-15:
The test with the vm wasn't successful. I couldn't use the serial port.
But a restart of my Windows and using another USB port works better.

The files get synced but I'm not able to call the `monitor` command.
I get now the error:
TypeError: cannot read property 'setType' of undefined
           at [anon] internal
           at [anon] (/index.js:5) strict preventsyield

BergenSoft, 2019-09-15:
I flashed the board again, and now everything works.
I could also use the `monitor` command.

closed 2019-09-15 by BergenSoft

neoniousTR, 2019-09-21:
I know this issue is closed, but just want to let you know... Fixed the following three things with the now published lowsync version 1.1.1:

- I had to call py -m pip install pyserial instead of pip install pyserial which should be reflected in the text lowsync outputs
- When flashing I had to press Enter sometimes
- It often says that it is "Testing the connection".. Working none-the-less. Maybe the timeouts are to small