neonious/lowjs #67: Can not flash my ESP32

GeneArnold, 2019-09-16:
I've ordered one of the NEONIOUS ONE boards but wanted to play around a little with the ESP32 board that I have till the new one arrives. The board I have does meet the specs and when I try to flash it I'm getting a time out error. I have GPIO0 set low and it the flasher does see the board on the port. Any tricks to help me out?

Here is the board I have-
https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Thank you

BergenSoft, 2019-09-16:
See https://www.lowjs.org/supported-hardware.html and read the last sentence.
The ESP-**WROOM**-32 Board is not supported.

GeneArnold, 2019-09-16:
My mistake, I thought I had the correct board. Would it be safe to say that this one would work? It does have the required amount of RAM? I'm excited to work with the NEONIOUS ONE when it comes in but if I need to create a few projects I'd like to know that I can also pull in some less expensive options. 

https://www.amazon.com/ESP32-WROVER-Development-Bluetooth-Consumption/dp/B07G2RBL37/ref=sr_1_3?crid=3017BKDUG4I9M&keywords=esp32+wrover&qid=1568648979&s=electronics&sprefix=ESP32%2Celectronics%2C128&sr=1-3

BergenSoft, 2019-09-17:
I'm no one from the lowjs team, but regarding the documentation it should work with any ESP32-WROVER board.
I bought a ESP32-WROVER-B board and it worked.

So I would say, yes it should work.

neoniousTR, 2019-09-19:
Just FYI, monitoring the thread (actually already did since the first post, however BergenSoft answered faster than me...)

closed 2019-10-04 by neoniousTR