neonious/lowjs #68: Load existing C library for Arduino

BergenSoft, 2019-09-17:
For Aduinos many ready-to-use libraries are available.
Is it possible to use them with lowjs without changing the library's code?
It would be a great feature and would greatly expand the possibilities for lowjs.

neoniousTR, 2019-09-19:
Hi!
Yes, of course that would be a great feature.
We have a paying customer who wants to have the possibility to upload a compiled C library into the low.js file system and use it from low.js. He is paying, so this feature will definitly become part of low.js sometime in Q4 2019.
However, for Arduino libs to work without change, we would also have to implement the Arduino IDE. That is work of a month, which we cannot do on the side..
Thomas

neoniousTR, 2019-11-25:
You can now interface C/C++ libraries such as Arduino libraries with the low native API: https://www.neonious.com/lowjs/documentation/lowjs-native-api.html

The interface still is work, as we are not aware of Arduino, but this does help in many cases.

closed 2019-12-16 by neoniousTR