neonious/lowjs #75: Set settings over API

zhakhalov, 2019-10-04:
Hi,

I'm new to low.js and looking for a possibility to set settings to the device using HTTP API.
Looking through the code of `lowsync` and `common` repos I found that most of the commands are implemented using HTTP API calls but couldn't find any proofs of that in the `low.js` repo.

BR,
Viktor

neoniousTR, 2019-10-04:
So, the way to do is to build a HTTP server with your own HTTP commands in your JavaScript user program and then set the settings via require('lowsys').setSettings({...}); See the lowsys API here: https://www.lowjs.org/lowjs_for_esp32/module-lowsys.html

Does this help you?

zhakhalov, 2019-10-05:
Hey there,

I did some research and found that API for setting does exist. First, you have to login with device password and later you can use SessionID from the response as a header for the following HTTP requests. Also, from the code of `lowsync` repo, I found APIs for `SetSettings` and `ShowSettings`. Seems API uses SessionID for authorization purposes.

The only question I have now is - how to unset settings and is it possible at all?

BR,
Viktor

neoniousTR, 2019-10-07:
I would strongly suggest to go the way I suggested as the API you are talking about is not documented and may change.

What settings do you want to unset in particular?

zhakhalov, 2019-10-17:
Hey there

I understand about using the suggested approach. Either way it would be really nice to have such stuff documented. Having access to the built-in API is way better the implementing the features that take memory and disk space on such resource-restricted platforms.

Speaking about __unsetting__ of the settings. As shown in the documents I tried to set custom field, in my case `firebase.projectId`. But with the evolution of the application, some properties become unused and need to be removed from the settings.

Also I would like to ask if it is possible to set the setting by wire. In case if I move from wi-fi to wi-fi I would like to have the possibility to change the settings by wire. Another case is work from PC, without wifi at all. There's no possibility to connect the device to the wifi in this case.

Best regards,
Viktor

neoniousTR, 2019-10-29:
Well, using the lowjs settings object for your own settings is not really a good idea. Where is that shown in the documents? I would simply save these settings somewhere else. Then you have the full control over them. Everything else is just hacking low.js to take advantage of the existing HTTPs protocol, which you can do, but which we cannot support.

You can change the low.js settings with any interface, you just need to write a JavaScript user program which reads this interface.

neoniousTR, 2019-11-19:
The newest low.js firmware now forbids using the settings object for other things, any key which is not known is rejected. Please see my comment above why.

closed 2019-11-19 by neoniousTR