neonious/lowjs #76: Build ESP32 firmware

zhakhalov, 2019-10-04:
Hi,

I'm new to low.js and looking for possibility to build firmware for the IoT device. I see that it is possible to build the engine for Windows/Linux/MacOS but couldn't find any clues on how to make the firmware for the ESP32 board. 

Also, I found that `lowsync` calls the neonious.com and sends device's MAC to it. I can make an assumption that firmware is device-restricted somehow.

BR,
Viktor

neoniousTR, 2019-10-04:
low.js is prebuilt on the neonious servers, and yes, it is device-restricted, however these are internals. You can flash low.js on any device via lowsync flash command for free.

Why do you want to build your own firmware?

zhakhalov, 2019-10-05:
Hey there,

I would like to add some functionality written in C. 
And also have some functionality being embeded into firmware to protect it from being rewritten

neoniousTR, 2019-10-07:
You have 3 options:

(1) If you are adding functionality which makes sense for every platform: Add the functionality to the Open Source low.js repo via Pull Request. It will be part of the ESP32 version then too.

(2) Within the next 3 months we will add the option that you can import dynamic libraries written in C in your JavaScript code. You just have to wait. However, this will still be basic for now.

(3) You ask us to implement your feature. We are experts and thus fast. So it can be a competitive option for you.

Thomas

neoniousTR, 2019-10-07:
Regarding "And also have some functionality being embeded into firmware to protect it from being rewritten":
That's what the user password is for. If set, nobody can change to content of the FS. Second, in production, I would turn of the IDE completly by disabling the HTTP and HTTPS ports.

zhakhalov, 2019-10-07:
Hi,

In other words, ESP32 specifuc parts of the project are propeietary. 

Honestly, first option makes little sense because I just cannot test the functionality on the real device without being able to build firmware)

I'm not sure I understand that no one can change files on the filesystem correct. Isn't it possible to download JavaScript file and run it?

Also, I would like to remove IDE from production build at all. This will save RAM and disk space.

BR,
Viktor

rtech-ViktorZhakhalov, 2019-10-13:
Thank you for the response.

neoniousTR, 2019-11-19:
The newest low.js now supports native modules! Please see https://www.github.com/neonious/low_native_api for an example.
Official documentation will follow soon.

neoniousTR, 2019-11-25:
You can find the official documentation of the low native API here: https://www.neonious.com/lowjs/documentation/lowjs-native-api.html

closed 2019-12-16 by neoniousTR