neonious/lowjs #77: [Question] lowsync flash throws a

mbariola, 2019-10-10:
Hi,
just installed lowsync. npm and node are both updated. 
When calling lowsync flash COM3 --init on win10 from a cmd with admin rights, I get the following error

```
An unexpected error has occured. The module '\\?\C:\Users\Massi\AppData\Roaming\npm\node_modules\lowsync\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`). Error: The module '\\?\C:\Users\Massi\AppData\Roaming\npm\node_modules\lowsync\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:805:18)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at bindings (C:\Users\Massi\AppData\Roaming\npm\node_modules\lowsync\node_modules\bindings\bindings.js:112:48)
    at Object.<anonymous> (C:\Users\Massi\AppData\Roaming\npm\node_modules\lowsync\node_modules\@serialport\bindings\lib\win32.js:1:36)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
```

Any idea on how to fix it? thanks

neoniousTR, 2019-10-11:
serialport is a sub-dependency, so I have to guess..

Did you update node after installing lowsync?
According to this: https://stackoverflow.com/questions/46384591/node-was-compiled-against-a-different-node-js-version-using-node-module-versio/52796884
this could be the problem.

Can you try to remove the npm package: npm uninstall -g lowsync
maybe then also do a npm cache clear
and reinstalling it?

Thank you

Thomas

neoniousTR, 2019-11-04:
Closing because of no activity.

closed 2019-11-04 by neoniousTR