neonious/lowjs #78: Passing arguments to the script

jirutka, 2019-10-24:
`low` does not allow to pass arguments to the script.

```sh
$ ./bin/low script.js --help
Syntax: ./bin/low [file.js]
```

The same for `./bin/low -- --help`.

neoniousTR, 2019-10-24:
You are right... This should be changed.
With the new commit https://github.com/neonious/lowjs/commit/9eb7dde9395743964ece6a4a1f0740d388f9614b low behaves just like node and also implements process.argv, process.argv0 and process.execPath.
New downloads should be available in a few hours.

jirutka, 2019-10-24:
Oh, that’s great, thank you a lot! :smile_cat: I’ve just started reading lowjs’s source code and thinking how to implement this and you already did it! :+1:

neoniousTR, 2019-10-24:
Some tasks are just fun..

But, if you want to contribute anywhere else, that would be great for the project.

jirutka, 2019-10-25:
I’ve just tried it and it works fine! :+1:

closed 2019-10-25 by jirutka