neonious/lowjs #81: A lot of tests are failing (on Travis CI)

jirutka, 2019-10-25:
I’ve tried to run the tests (`npm run test -- run`) on Travis CI (see [.travis.yml](https://github.com/jirutka/lowjs/blob/8fba5c8970c496e0a7983c6e082d8d0b349720e6/.travis.yml)) and I’m quite terrified that **a lot of tests failed** and it eventually got stuck on test 2021/2107, so it even didn’t finish. You can see the log [here](https://travis-ci.com/jirutka/lowjs/builds/133568090).

Why is it so? Is there some problem with the versions of some system libs or something?

neoniousTR, 2019-10-25:
The tests are the tests of Node v10. So naturally, as we are in the process of implementing the rest of Node v10 (see https://www.neonious.com/lowjs/documentation/nodejs-api.html), many tests are still failing.

So this is definitly a place where you can find small tasks which can be very rewarding.

However, I think something else is wrong here, as as far as I see it ALL tests are failing.

Did you try it exactly as written here:
https://github.com/neonious/lowjs/blob/master/test/README.md

Including doing the prerequisites? If done correctly you should get a summary html at the end.

jirutka, 2019-10-25:
> The tests are the tests of Node v10. So naturally, as we are in the process of implementing the rest of Node v10 (see https://www.neonious.com/lowjs/documentation/nodejs-api.html), many tests are still failing.

Aha, I see now!

> Including doing the prerequisites?

Yes, I did, as you can see in the [.travis.yml](https://github.com/jirutka/lowjs/blob/8a6f347b8b0c6a44a4a3cda979271ad96bb63fea/.travis.yml#L10-L11) file

> If done correctly you should get a summary html at the end.

As I said, it doesn’t reach the end because one of the tests hangs.

neoniousTR, 2019-10-26:
I have an exact list of things to do till mid November. This includes implementing a full first version of a native API so a mix of C++ and JavaScript can be programmed even on the microcontroller, and taking a look at some other issues people reported. Then I will check why the tests completly fail and when that works again, that will become 1.0.0 - the first tag

neoniousTR, 2019-11-04:
In the master version the tests script works again.
Still, many many tests fail. Many because they are not really correct and even fail with Node.JS (see NODE_FAILED on the test info). Many because they test things which are not and will not be implemented with low.js (cluster, http2, etc.). Need to clean up here.
But still, many which fail are now good targets to tackle!

jirutka, 2019-12-25:
The test suite still looks pretty much broken (but at least doesn’t hang now):

```
238 / 2117 tests OK.
```

Full log: https://api.travis-ci.com/v3/job/270329926/log.txt

neoniousTR, 2020-01-29:
Just checked.. Many, many tests fail because low.js simply outputs stack traces a bit different... and so on.

So, most tests simply fail because they have to be adapted. Will work on it in the future.