neonious/lowjs #82: Git tags for releases

jirutka, 2019-10-25:
Can you please start creating git tags for the releases? I want to create an Alpine Linux package for low.js and would like to integrate low.js with [pkg](https://github.com/zeit/pkg); absence of tags is quite an obstacle.

neoniousTR, 2019-10-26:
I have an exact list of things to do till mid November. This includes implementing a full first version of a native API so a mix of C++ and JavaScript can be programmed even on the microcontroller, and taking a look at some other issues people reported. Then I will check why the tests completly fail and when that works again, that will become 1.0.0 - the first tag

neoniousTR, 2019-11-19:
We have the first git tag with the current release, which has lots of changes (see Change log)! It is v1.3.0, as in the change log we already have 1.0.0. Hope this helps?

closed 2019-12-16 by neoniousTR

jirutka, 2019-12-25:
Thanks! Today I created Alpine package for lowjs ([source](https://github.com/alpinelinux/aports/tree/master/testing/lowjs)) and pushed it into the Alpine’s testing repository (see [here](https://pkgs.alpinelinux.org/packages?name=lowjs&branch=edge)).