neonious/lowjs #83: Size problem with package manager

ygageot, 2019-11-24:
It is not evident to install package with a limited storage.

Is it possible to display :
- the size of the installed packages ?
- the size of a candidate package ?

It can be useful when you try to install an additional package and you have not a lot of storage.

neoniousTR, 2019-11-25:
This is a good feature request.

We will not be able to show the size of a package before you actually try to install it, because we do not know how many other packages are required for the package till we try installing (these are then also installed, but are hidden. Packages which are no longer needed by the selected packages are always uninstalled).

But we can show the size of the packages currently installed and give a better feedback if the size of the package (+subpackages being installed right now is too large (and how much space has to be freed for the next try).

I am adding this to the todo list. But, to be honest, I have no idea when we can do this, we have other, paid work, which is filling our agenda.

ygageot, 2019-11-27:
Thanks for your message.

In fact I would like to transfer a 4600 lines Nodejs personnel project + 7 several npm packages.

With the basic Neonious One hardware, it is not possible to load all the external packages.

I plan to replace the flash memory with W25Q128JV.

On the Digit-Key web site it is difficult to find the good hardware because it exists many packagings.

I found the W25Q128JVSIQ TR packaging, Is it a good choice  ?

If I understand the 64 chip contains  8Mb = 4MB for IDE +   4Mb for user space
the 128 is                                        16MB = 4MB for IDE + 12Mb for user space
Is it good ?

Thanks in advance for my 2 questions

Yves

NB : my project is for a small autonomous sailing boat. 
Actually it is with a PI for the main brain in Nodejs and  2 peripheral ESP32 C++ for sensors and servos using deep-sleep.
The power consumption of the PI is too important and difficult to manage.
It is the reason why I want to try the Neonious One with low.js
I prefer to use flash memory in place of SD for liability.

Yves

neoniousTR, 2019-11-27:
Yes, that is the right chip, and your calculation also is correct.

Thomas

ygageot, 2019-11-27:
Thanks for your message.In fact I would like to transfer a 4600 lines Nodejs personnel project + 7 several npm packages.With the basic Neonious One hardware, it is not possible to load all the external packages.I plan to replace the flash memory with W25Q128JV.On the Digit-Key web site it is difficult to find the good hardware because it exists many packagings.I found the W25Q128JVSIQ TR packaging, Is it a good choice  ?If I understand the 64 chip contains  8Mb = 4MB for IDE +   4Mb for user spacethe 128 is                                        16MB = 4MB for IDE + 12Mb for user spaceIs it good ?Thanks in advance for my 2 questionsYvesNB : my project is for a small autonomous sailing boat. Actually it is with a PI for the main brain in Nodejs and  2 peripheral ESP32 C++ for sensors and servos using deep-sleep.The power consumption of the PI is too important and difficult to manage.It is the reason why I want to try the Neoniuos One with low.jsI prefer to use flash memory in place of SD for liability.

closed 2020-05-15 by neoniousTR