neonious/lowjs #84: m5stack

0x77dev, 2019-12-02:
m5stack support?

neoniousTR, 2019-12-03:
Depends on whether it uses the ESP32-WROVER or external 4 MB PSRAM.

Just checked: The M5Camera does. The other boards, I do not think so.

0x77dev, 2019-12-03:
I have m5stack Core, I will test it out, maybe we can write some lib for m5stack?

neoniousTR, 2019-12-15:
Try it, but I really think m5stack Core does not have the PSRAM.

0x77dev, 2019-12-20:
Its works^ but... after 5seconds it crashes and goes to infinity reboot.

neoniousTR, 2020-01-15:
I think it does not have the PSRAM. Am I correct?

closed 2020-01-29 by neoniousTR

tobozo, 2020-02-15:
M5Stack-Fire has psram, same for Odroid-Go.

airtonix, 2021-02-22:
m5paper has psram too. 

[image]