neonious/lowjs #85: Can't upload large JS file, also --no-transpile doesn't work

pm4r, 2019-12-05:
I'm trying to sync project containing single JS file, size is ~95KB, however lowsync fails with "The device cannot be reached with the provided protocol, IP and port". After running lowsync, cosole output looks like this:

`
Fetching file system listings...
Reminder: No password set! Please set one via the lowsync settings set web.password="..." command. Continuing...
Synchronizing... |░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░| 0% | main.js
`

Then after 10-15 seconds:

`
Testing connection to microcontroller at https://192.168.100.28:8443... This can take a while if your connection is bad. If the url is incorrect, please abort lowsync and change the config file or run lowsync.
`

And after another 1-2 minutes:

`
The device cannot be reached with the provided protocol, IP and port (192.168.100.28 8443 ssl: true).? 
What is the IP address of the microcontroller on your network? (192.168.0.1) 
`

For shorter files it syncs without problems, however usually it prints `Testing connection to microcontroller...` message but finishes sucessfully.

Same file renamed to .txt uploads without issue, so it affects only js files.
I tried to estimate the limit by creating dummy js file with different sizes, and successfully synced with up to about 80KB in single file.

I thought it might be because code is being transpiled, and assumed it gives additional overhead, so (because my file is ES5) I tried:
`lowsync --no-transpile`
but then it prints:

`
lowsync [sync]

Sync and, by default, transpile the files from this computer to the
microcontroller.

Options:
  --help          Show help                                            [boolean]
  --version       Show version number                                  [boolean]
  --no-transpile  Disable the transpilation of source files (only >=ES6
                  JavaScript files, NO TypeScript, etc.) to ES5. Be sure that
                  you know what you are doing before using this option![boolean]
  --restart       Enable/disable (re)starting of the program if the filesystem
                  on the microcontroller has changed. Optionally you may append
                  =<true|false> to this option.                        [boolean]
  --monitor       Enable/disable monitoring the program after sync. Implies
                  --restart. Optionally you may append =<true|false> to this
                  option.                                              [boolean]
Unknown argument: transpile
`

So it looks like this option is not recognized (also with `lowsync sync --no-transpile`).

Then I added in `lowsync.config.json` `"transpile": false` but behavior doesn't change. It looks like it might be ignored because simple led fading example also works but I expected it shouldn't (because of `let` keyword).

I'm trying to figure out if low.js can run Clojurescript which is lisp dialect compiled into Javascript (like node.js on ESP32 wasn't cool enaugh ;)), and 95 KB is just simple Hello world printed on the console. 

My hardware is ttgo T18 v3.0. Lowsync version is 1.1.1.

neoniousTR, 2019-12-05:
OK sounds serious. Let me try to fix both problems within the next 48 hours...

neoniousTR, 2019-12-07:
I found both bugs and fixed them. We are also currently working on a larger release, which we plan on get finished middle of next week. Is it OK if we release the fixes with this new release?

pm4r, 2019-12-07:
Nice, thanks. Sure, I will wait then and let you know after I test the new release.

neoniousTR, 2019-12-15:
Please try out the new version 1.4.0 (requires lowsync 1.4.0), see: https://neonious.com/documentation/changelog

pm4r, 2019-12-16:
Ok, now it's uploaded quickly and successfully. Even simple hello world from Clojurescript worked from first kick :). I'm going to test it more in the next 2 weeks and let you know if there are any issues.

The one thing I had to change was:
```
lowsync flash /dev/ttyUSB0 --init
```
to
```
lowsync flash --port=/dev/ttyUSB0 --init
```
The former gave me 

> Error: No port specified. Please use --port=.. to specify the port.

But worked last time I checked.

neoniousTR, 2019-12-16:
That is correct, format changed with lowsync 1.4.0 because the parameter now is a firmware image (optional).

OK keeping this issue open till you have more feedback in the next 2 weeks.

closed 2020-01-28 by neoniousTR