neonious/lowjs #86: Problem with setInterval()

gizmodus, 2019-12-22:
Hello! First, thanks a lot for this great library. It's really cool to control an ESP32 with Javascript.

I experimented a bit with the library and experienced some problems with setInterval(). This is my code which works fine:
```
setInterval(() => {
  console.log('Message - ' + new Date());
}, 500);
```

However, if I increase the interval from 500ms to 5 seconds, the console is not printing anything anymore. The funny thing is, that when I press enter in the console, the messages appear. I don't think the problem is related to the transmission between ESP32 and PC but rather something in the microprocessor. If instead of logging something to the console, I turn an LED on and off, I experience the same problems (nothing is happening, until I press enter in the console).

Do you have an idea what the reason for this could be?

neoniousTR, 2019-12-24:
Fixed, please update! See https://neonious.com/documentation/changelog

gizmodus, 2019-12-26:
@neoniousTR Wow, thanks a lot for the quick fix! I tested it and everything seams to work now. Thank you!

closed 2019-12-26 by gizmodus