neonious/lowjs #88: Socketcluster-client

johancoppens, 2020-01-15:
Today I had my first experiences with lowjs and I was very excited! I wanted to use the powerful socketcluster.js framework. Therefore I installed the [socketcluster-client](https://github.com/SocketCluster/socketcluster-client/blob/master/lib/scclientsocket.js) npm package on my ESP32-WROVER-KIT. I installed it with "lowsync install socketcluster-client". The install works and i can import it in my code like this:
```
const scc = require('socketcluster-client')
const options = {
    host: '192.168.0.224:8000'
  }
const socket = scc.create(options)
socket.on('error', (err) => {
    console.log(err)
})
let counter = 0
setInterval(function(){
    console.log('event', counter++)
    socket.emit('sampleClientEvent', counter)
}, 2000)
```
But I get these errors:
```
Hello world
{ Error
    at SocketProtocolError (module:/sc-errors/index.js:1) strict construct
    at [anon] (module:/socketcluster-client/lib/scclientsocket.js:1) strict
    at [anon] (module:/socketcluster-client/lib/scclientsocket.js:1) strict
    at [anon] (module:/component-emitter/index.js:1) strict
    at [anon] (module:/socketcluster-client/lib/sctransport.js:1) strict
    at [anon] (module:/socketcluster-client/lib/sctransport.js:1) strict
    at r (module:/ws/lib/event-target.js:1) strict
    at emit (lib:events:158) strict
    at [anon] (module:/ws/lib/websocket.js:1) strict
  name: 'SocketProtocolError',
  message: 'Socket hung up',
  code: 1006 }
```
Any idea if this can be solved?
The code works on node.js on my local PC.

Thanks in advance!

neoniousTR, 2020-01-15:
Well, Node.JS and low.js are still a little bit different in the bits and pieces and this sometimes still breaks some libs.

Does the code work if you use the low.js binary with the transpile option? Just download low.js and replace node .. with <pathtolow>/bin/low --transpile ..

Note: --transpile is slow, and just for testing!

It most probably will not, which is good because then both architectures behave the same and then the problem can be easily debugged on the PC.
Can you try to build me such a zip file with your .js file and node_modules which I can call with low on PC to debug quickly?

johancoppens, 2020-01-16:
Hello
Thanks for the quick response! 
Hereby the requested zip file. The server can be started with the command 'node server.js' in the server directory. The client is in the client directory. When I run the client with 'node index.js', everything runs fine, I have a connection with the server and the messages are send successfully. When launched with './low --transpile index.js' on my local PC, I get the same errors as on the device.
In the meantime I have ordered a NEONIOUS ONE, I look forward to receiving it ;-)
[test_low_socktetcluster.zip](https://github.com/neonious/lowjs/files/4069915/test_low_socktetcluster.zip)

johancoppens, 2020-01-29:
Thanks for fixing this! I have build low based on the latest git version on my local PC and when I run a client it works. But how do I get the latest version on my neonious one? Seems 'lowsync update install' does not update to the latest git version?

neoniousTR, 2020-01-29:
Can you retry now? It should update to 20200129 which does include the fix.
If it is not 20200129, what does the neonious one update to?

johancoppens, 2020-01-31:
The neonious is (and I think was) updated  to 20200129. The cause was a main update of socketcluster-client a couple of days ago which was no longer compatible with my local version. I have managed to install the older version with lowsync install socketcluster-client@14.3.1 on the neonious. I had to try this several times, but after +- 50% I received an error many times: "An error has occurred. Please make sure the microcontroller has access to the Internet." In the end it seems that the package was installed in the background succesfully. Installing an older version was not possible via the package manager in the web interface, or am I wrong? I will test it the next couple of days (also the newer version). It seems very promising: websockets on steroids with pub/sub and autoreconnect on the esp32 ;-).

neoniousTR, 2020-01-31:
The IDE package manager interface does not support versioning, that is correct. So it was a good idea to try through lowsync.

I will add the problem with the lowsync command to my list of things to look at.

Till then, I am interested to hear whether the package is installed and works, even though the error was shown.

commit 94f8e691 (2020-01-27): fixed #88