neonious/lowjs #91: Make error

fc92, 2020-01-19:
I just tried to build from source on x86_64 Linux and I got the following cast error. 

> $ make
cd deps/open62541 && rm -rf build && mkdir build
cd deps/open62541/build && cmake ..
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /home/f/.espressif/python_env/idf3.3_py2.7_env/bin/python (found version "2.7.17") 
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- The selected architecture is: posix
-- Performing Test CC_HAS_-fstack-protector-strong
-- Performing Test CC_HAS_-fstack-protector-strong - Success
-- Performing Test CC_HAS_-fstack-clash-protection
-- Performing Test CC_HAS_-fstack-clash-protection - Success
-- Performing Test CC_HAS_-mcet -fcf-protection
-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/f/lowjs.official/deps/open62541/build
cd deps/open62541/build && make
make[1] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[2] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
Scanning dependencies of target open62541-generator-types
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  1%] Generating src_generated/open62541/types_generated.c, src_generated/open62541/types_generated.h, src_generated/open62541/types_generated_handling.h, src_generated/open62541/types_generated_encoding_binary.h
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  1%] Built target open62541-generator-types
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
Scanning dependencies of target open62541-generator-namespace
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  3%] Generating src_generated/open62541/namespace0_generated.c, src_generated/open62541/namespace0_generated.h
INFO:__main__:Preprocessing /home/f/lowjs.official/deps/open62541/tools/schema/Opc.Ua.NodeSet2.Minimal.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  3%] Built target open62541-generator-namespace
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
Scanning dependencies of target open62541-generator-statuscode
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  5%] Generating src_generated/open62541/nodeids.h
[  5%] Generating src_generated/open62541/statuscodes.h, src_generated/open62541/statuscodes.c
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  5%] Built target open62541-generator-statuscode
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
Scanning dependencies of target open62541-generator-transport
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  7%] Generating src_generated/open62541/transport_generated.c, src_generated/open62541/transport_generated.h, src_generated/open62541/transport_generated_handling.h, src_generated/open62541/transport_generated_encoding_binary.h
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  7%] Built target open62541-generator-transport
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
Scanning dependencies of target open62541-object
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[3] : on entre dans le répertoire « /home/f/lowjs.official/deps/open62541/build »
[  9%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
[  9%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o
[ 11%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/types_generated.c.o
[ 13%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/transport_generated.c.o
[ 13%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/statuscodes.c.o
[ 15%] Building C object CMakeFiles/open62541-object.dir/src/ua_util.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src/ua_workqueue.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src/ua_timer.c.o
[ 19%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o
[ 21%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o
[ 23%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_session.c.o
[ 23%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o
[ 25%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o
/home/f/lowjs.official/deps/open62541/src/server/ua_server.c: In function ‘UA_Server_addTimedCallback’:
/home/f/lowjs.official/deps/open62541/src/server/ua_server.c:287:38: error: cast between incompatible function types from ‘UA_ServerCallback’ {aka ‘void (*)(struct UA_Server *, void *)’} to ‘UA_StatusCode (*)(void *, void *)’ {aka ‘unsigned int (*)(void *, void *)’} [-Werror=cast-function-type]
  287 |                                      (UA_ApplicationCallback)callback,
      |                                      ^
/home/f/lowjs.official/deps/open62541/src/server/ua_server.c: In function ‘UA_Server_addRepeatedCallback’:
/home/f/lowjs.official/deps/open62541/src/server/ua_server.c:296:41: error: cast between incompatible function types from ‘UA_ServerCallback’ {aka ‘void (*)(struct UA_Server *, void *)’} to ‘UA_StatusCode (*)(void *, void *)’ {aka ‘unsigned int (*)(void *, void *)’} [-Werror=cast-function-type]
  296 |                                         (UA_ApplicationCallback)callback,
      |                                         ^
/home/f/lowjs.official/deps/open62541/src/server/ua_server.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-static-in-inline’ [-Werror]
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/open62541-object.dir/build.make:283: CMakeFiles/open62541-object.dir/src/server/ua_server.c.o] Error 1
make[3] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[2]: *** [CMakeFiles/Makefile2:439: CMakeFiles/open62541-object.dir/all] Error 2
make[2] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make[1]: *** [Makefile:152: all] Error 2
make[1] : on quitte le répertoire « /home/f/lowjs.official/deps/open62541/build »
make: *** [Makefile:127: deps/open62541/build/bin/libopen62541.a] Error 2

neoniousTR, 2020-01-25:
GCC 8 adds this new check which breaks things. I am not using GCC 8, so I did not see this problem.

I think I fixed it (set the error to ignore, because all is fine the way it is), but I cannot check without GCC 8.

Can you update the GIT and the git submodules (deps/open62541) and try to recompile now?

fc92, 2020-01-25:
I tried but there is an usupported compiler settings in the CMakeLists.txt file so I commented it:
`#                    -Wno-static-in-inline # clang doesn't like the use of static inline methods inside static inline methods`
Now I have a linker error:

> mkdir -p bin
g++ -o bin/low deps/mbedtls/library/*.o deps/mbedtls/crypto/library/*.o deps/c-ares/libcares_la-*.o deps/duktape/src-low/duktape.o src/low_main.o src/low_module.o src/low_native.o src/low_native_aux.o src/low_process.o src/low_loop.o src/low_fs.o src/low_fs_misc.o src/low_http.o src/low_net.o src/low_dgram.o src/LowDatagram.o src/low_tls.o src/low_dns.o src/low_crypto.o src/LowCryptoHash.o src/low_data_thread.o src/low_web_thread.o src/low_alloc.o src/low_system.o src/LowFile.o src/LowFSMisc.o src/LowServerSocket.o src/LowSocket.o src/LowHTTPDirect.o src/LowSignalHandler.o src/LowDNSWorker.o src/LowDNSResolver.o src/LowTLSContext.o src/low_native_api.o src/low_opcua.o app/main.o app/transpile.o -O3 -DLOW_VERSION="\"`git show -s --format=%cd --date=format:%Y%m%d`_`git rev-parse --short HEAD`\"" -lm -ldl -lpthread deps/open62541/build/bin/libopen62541.a
/usr/bin/ld: deps/open62541/build/bin/libopen62541.a: error adding symbols: file format not recognized
collect2: error: ld returned 1 exit status
make: *** [Makefile:59: bin/low] Error 1

Can you add more precision regarding your standard configuration? (I compiled with clang and also faced a linker issue. I tried to use the ldd linker without success.)

neoniousTR, 2020-01-27:
I reverted the fix because it breaks other compilation systems.

low.js compiles well under Mac, Debian and Alpine Linux (all with standard gcc compiler). Never checked other systems.

Can you please give an exact info on what system you are using? The easiest path forward is me trying to reproduce this.

johancoppens, 2020-01-29:
I had the same problems building low.js on ubuntu. After selecting gcc-7 and g++-7 after following this guide [https://linuxconfig.org/how-to-switch-between-multiple-gcc-and-g-compiler-versions-on-ubuntu-20-04-lts-focal-fossa](https://linuxconfig.org/how-to-switch-between-multiple-gcc-and-g-compiler-versions-on-ubuntu-20-04-lts-focal-fossa) it builds successful. Hope it helps!

fc92, 2020-01-29:
Thank you @johancoppens. I confirm using your solution solved my compilation issue!

closed 2020-01-29 by fc92