neonious/lowjs #92: Error: Cannot find module 'js-levenshtein' when lowsync

ipajonro, 2020-01-24:
`Fetching file system listings...
Preparing synchronization...0%{file}Unexpected error: Cannot find module 'js-levenshtein' Error: Cannot find module 'js-levenshtein'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.js-levenshtein`

lowsync version: 1.4.4
npm version: 6.13.4
nodejs version: 10.18.1

I can't sync my ESP32 after flashing it because of this error

neoniousTR, 2020-01-25:
This is interesting! Can you install the package globally: npm i -g --unsafe-perm js-levenshtein
and then see if it works. Maybe some other package is also missing.

If that's it, I will add the package to the package.json in the next release.

ipajonro, 2020-01-25:
It works! Now I can compile with ES6 transpilation, and it also works with NodeJS 12.14.1.

Thanks so much 😄

closed 2020-01-25 by ipajonro

neoniousTR, 2020-01-29:
The package is now part of lowsync 1.4.5 which was just published on npm (might take some hours to show).