neonious/lowjs #93: mqtt module cannot be used

yulin724, 2020-01-28:
mqtt module cannot be used.

var ws = require('ws')
console.log(ws)

var mqtt = require('mqtt')
console.log(mqtt)

ws module is copied from char server example.

mqtt is from "npm/yarn install mqtt".

when I require this mqtt module, i got"

../bin/low heroServer.js 
SyntaxError: parse error (line 31)
    at [anon] (/Users/yulin724/dev/lowjs/hero/node_modules/ws/lib/WebSocket.js:31) internal
    at [anon] (deps/duktape/src-low/duktape.c:71379) internal
    at require () native strict preventsyield
    at [anon] (/Users/yulin724/dev/lowjs/hero/node_modules/ws/index.js:9) strict preventsyield
    at require () native strict preventsyield
    at [anon] (/Users/yulin724/dev/lowjs/hero/heroServer.js:6) preventsyield

neoniousTR, 2020-01-28:
low.js supports ES 5.1, not ES 6+. This is not a problem on the microcontroller, as both the IDE and lowsync transpile while uploading, so the limitation is not noticed. But on PC this limitation is real. You could use low --transpile heroServer.js , but this is slow and only for debugging.

ipajonro, 2020-02-15:
I'm currently having same problem with Axios:

```
import { get } from 'axios';

get(`${url}`)
.then(response => {
  console.log(response);
})
.catch(error => {
  console.log(error);
});
```

```
ReferenceError: identifier 'discard2headers' undefined
    at [anon] internal
    at [anon] (lib:internal/http:683) strict preventsyield
Fatal error, restarting program...
```

And also:

```
const axios = require('axios');

axios.get(`${url}`)
.then(response => {
  console.log(response);
})
.catch(error => {
  console.log(error);
});
```

```
ReferenceError: identifier 'discard2headers' undefined
    at [anon] internal
    at [anon] (lib:internal/http:683) strict preventsyield
Fatal error, restarting program...
```

I'm using LowJS 1.4.5 (Latest).

neoniousTR, 2020-02-16:
This is not the same issue as with mqtt. But, as you opened a new issue, lets continue this in the new issue #94 !

closed 2020-03-24 by neoniousTR