neonious/lowjs #94: Axios module fails

ipajonro, 2020-02-16:
I'm currently notice a problem with Axios:

```
const axios = require('axios');

axios.get(`${url}`)
.then(function(response) {
  console.log(response);
})
.catch(function(error) {
  console.log(error);
});
```

```
ReferenceError: identifier 'discard2headers' undefined
    at [anon] internal
    at [anon] (lib:internal/http:683) strict preventsyield
Fatal error, restarting program...
```

I'm using LowJS 1.4.5 (Latest).

When **url** parameter is like **https://firestore.googleapis.com/v1/projects/firebase-id/databases/(default)/documents/devices** it fails, but when it's like **https://maker.ifttt.com/trigger/webhook/with/key/ifttt_key** it doesn't... 
Same code on a regular Node app works, also with Postman (testing both url with Get method). Any idea?

neoniousTR, 2020-02-16:
Can you try in the newest version? Just fixed this, I think.

ipajonro, 2020-02-16:
I'm using LowJS 1.4.5. I have seen in ChangeLog versions 1.4.5.1 and 1.4.5.2 but they aren't in npm

neoniousTR, 2020-02-16:
low.js is never in npm, only the flashing tool lowsync

It does not have a version number yet, because it is just a small change.

If you are using a PC, the downloadable packages are updated: https://www.neonious.com/lowjs/downloads.html
If you are using a microcontroller, lowsync flash will flashed an updated version

ipajonro, 2020-02-16:
Sorry for that, my mistake.

I just flashed again my ESP32 and error persists, although line error now is in 682:

```
ReferenceError: identifier 'discard2headers' undefined
    at [anon] internal
    at [anon] (lib:internal/http:682) strict preventsyield
Fatal error, restarting program...
```

neoniousTR, 2020-02-16:
I am sorry for not fixing well. I tested the fix, but most probably took a domain which worked all the time.

This will be fixed within the next 2 days.

neoniousTR, 2020-02-19:
This should now work in the newest version. Please retry!

ipajonro, 2020-02-19:
Now it works perfectly, thanks a lot! 😄

commit 6e76e2d3 (2020-02-19): fixed #94
closed 2020-02-19 by ipajonro