neonious/lowjs #95: unable flash to ESP32-Devkit-C

waninlezu, 2020-02-22:
dears, 

I'm sure I have the ESP32-WROVER-B (4M flash and 8M PSRAM) with CP2012N USB to UART
but got below error when trying to flash, any ideas?

```
$ lowsync flash --port=/dev/cu.SLAB_USBtoUART --init

*** Step 1/3: Probing ESP32 microcontroller
    now checking if it is an ESP32-WROVER... (takes a while)
Error: ESP32 is not an ESP32-WROVER or at least does not have required 4 MB PSRAM!
Please check: https://www.lowjs.org/supported-hardware.html
```

Chip info:

```
$ esptool.py   --port=/dev/cu.SLAB_USBtoUART flash_id
esptool.py v3.0-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 84:0d:8e:e3:aa:fc
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

> It seems to flash the small program which checks whether the board is a WROVER. This program does not function, however, or so it seems. After it no longer responds, can you check what the program outputs with any terminal program? Just connect to the Serial Port and see what it outputs. It should either output NOT_WROVER or IS_WROVER in a loop, if it does not and outputs something else, this would help me.

the small program does not function and output as:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5212
load:0x40078000,len:9036
ho 0 tail 12 room 4
load:0x40080400,len:6572
entry 0x400806f8
^Es0Tp!jhq      cv!bFtfQv!r0GG`<!KanAJaPra{GjB!rXbp4uegb11b1bb
```

neoniousTR, 2020-03-03:
Just FYI: This does not indicate the small program is broken. It works with another bitrate than the console. Not sure why, reverted that and updated the ESP-IDF version.

lowsync v1.4.7 was just published, please retry with the new version.

waninlezu, 2020-03-04:
thanks @neoniousTR 
It works for me now, thanks. great job!

closed 2020-03-04 by neoniousTR