neonious/lowjs #96: Can't install uart module

ipajonro, 2020-02-23:
I'm trying to install **uart module** via `lowsync install uart` and I'm getting an error. If I install other module like **gpio** it works fine.

[image]

Tested on **Lowsync 1.4.6**

neoniousTR, 2020-02-26:
Hello!
These are internal modules and are already available!
The npm modules have nothing to do with these modules. Installing gpio will introduce wrong modules to the system.

I understand how you can thing this is the right thing to do.
I will keep this issue open as a reminder for myself to add a warning in case somebody tries this.

Thomas

ipajonro, 2020-02-26:
Ahhh okey, I was testing **[uart_loop.js](https://github.com/neonious/lowjs_esp32_examples/blob/master/general/uart_loop.js)** and it seemed not to works, that is why I tried with **lowsync install**.
So, knowing that is not the way, **uart_loop.js** is not showing anything in console, when it should be, any idea?

Thanks

neoniousTR, 2020-02-26:
On what device are you trying this?

Are you running the plain unchained uart_loop.js?

Thank you,

Thomas

ipajonro, 2020-02-26:
I'm trying with an ESP32, in a new project I created and synced with lowsync.

I have copied uart_loop.js content and pasted in a new file called index.js. Without any change.

neoniousTR, 2020-02-27:
Fixed this in the newest firmware. Please reflash and retry!

ipajonro, 2020-02-27:
Now I'm getting an interrogation symbol as output, with original uart_loop.js.

```
Restarting program...
Starting monitor...
--- User program's output: ---
�
--- Program exited. ---
```

I tried with different encodings in `pipe1.setEncoding('utf8');` and `pipe2.setEncoding('utf8');` and nothing, same result.

neoniousTR, 2020-02-27:
You are right, I have one important change in my uart_loop.js: The order of the constructors. Fixed it in the repo, see commit changes: https://github.com/neonious/lowjs_esp32_examples/commit/e8ecbfb7a1345fac730141a05a6970243d401c4e
Also added a comment to explain why this is necessary.

ipajonro, 2020-02-27:
Thanks a lot! Also I tested with a real serial connection and works fine.
Is there any way to implement 'end' event?

```
pipeIn.on('end', () => {
  console.log('ONEND!');
});
```

Because in 'data' event I'm receiving my message cutted in a few parts, so I can know when start to save chunk and when to stop

neoniousTR, 2020-02-28:
That's not how UART works. There is no end indicator.
You have the same challenge as with tcp streams (net module of Node.JS): You need to create some protocol where you know where the end is. And then handle packets which come in.

Tip from my side: JSON lines is easy:
Simply always write JSON.stringify(object) + '\n'
And for reading you take the uart and put it through a readline object (also a native module of Node.JS), on every on('line') you simply JSON.parse

ipajonro, 2020-02-28:
Thanks you so much! It was very helpful.
For my part is all resolved, you can close this issue when you want (as you said you will keep it open as a reminder) 😄

closed 2020-05-15 by neoniousTR