neonious/lowjs #98: Bluetooth support

p610, 2020-03-08:
Does `lowjs` support Bluetooth? Did not see any mention of bluetooth interface in the documentation.

neoniousTR, 2020-03-24:
Hi,  I'm finally back from vacation!

low.js currently does not support Bluetooth.

Bluetooth support however is straightforward to implement, we already have the development plan to do this. However it will take a month of software development to do this, and we cannot spend this time for free. It is, in fact, the largest development which still has to be done with low.js.

So effectively, Bluetooth will be available as soon as a company which uses low.js needs it and is willing to sponsor the development.

wallaroo, 2021-01-04:
can i suggest to activate a crowdfunding initiative to let small businesses to share the funding these kind of key features?

ThomasRogg, 2021-01-06:
That could be an option. We are full with work however and thus beginning of Q2 2021 a crowdfunding could start. We will think about it in February.