neonious/lowjs #100: Error 'unhandled Promise reject'

ipajonro, 2020-03-29:
When I do `new Promise((resolve, reject) => {})` I'm getting the following error:

```
TypeError: unhandled Promise reject: cannot read property 'push' of undefined
    at [anon] internal
    at formatPromise (lib:util:1006) strict
    at formatValue (lib:util:690) strict tailcall
    at formatWithOptions (lib:util:212) strict tailcall
    at log (lib:console:161) strict
    at getResponsePacket (/GTXUtils.js:168) strict
    at _callee$ (/index.js:23) strict
    at tryCatch (lib:init:1163) strict
    at invoke (lib:init:1360) strict tailcall
    at tryCatch (lib:init:1163) strict
    at invoke (lib:init:1244) strict
    [...]
Fatal error, restarting program...
```

I think it may be caused because in lowjs/util.js function formatPromise has 'output' variable uninitialized (`let output = []`), and it is trying to push in an undefined object.

```
function formatPromise(ctx, value, recurseTimes, keys) {
    let output;
    /*
        const [state, result] = getPromiseDetails(value);
        if (state === kPending) {
            output = ['<pending>'];
        } else {
            const str = formatValue(ctx, result, recurseTimes);
            output = [state === kRejected ? `<rejected> ${str}` : str];
        }
    */
    for (var n = 0; n < keys.length; n++) {
        output.push(formatProperty(ctx, value, recurseTimes, keys[n], 0));
    }
    return output;
}
```

neoniousTR, 2020-03-29:
Good find! Partially fixed in git, it now no longer crashes, but it does not show the same great information as Node.JS does (pending vs resolved vs rejected). This is not a priority for me, so if you or anybody else wants this to work exactly as it is in Node, I am looking forward to a Pull Request.

Just as doc: Promises in low.js are implemented in /lib_js/init.js

ThomasRogg, 2020-09-15:
Reimplemented Promises in native C++ for speed, and fixed this along the way!

Thomass-MacBook-Pro:lowjs t.rogg$ bin/low 
> new Promise(function(resolve, reject) {});
[Promise] { <pending> }

closed 2020-09-15 by ThomasRogg