neonious/lowjs #101: setFrequency for PWM

Craigzyc, 2020-04-11:
This may just be a misunderstanding since the documentation isn't quite clear but setFrequency isn't working as I expect it to.

`gpio.pins[13].setType(gpio.OUTPUT).setFrequency(80).setValue(0.99); 
`
This throws error TypeError: undefined not callable (property 'setFrequency' of [object Object])

Changing that line to 
`gpio.pins[13].setType(gpio.OUTPUT).setValue(0.99); 
`
properly operates the pwm but I need it to be at 80Hz

neoniousTR, 2020-04-15:
setFrequency is currently global, for all GPIOs. This is because the second CPU on the neonious one does not support a different frequency for every pin.

So that would be:
gpio.setFrequency(80);
gpio.pins[13].setType(gpio.OUTPUT).setValue(0.99);

However, keeping this issue as a reminder to change this for all other boards.

Craigzyc, 2020-04-19:
That works. Thank you however, PWM is unusable currently due to instability. 
Checking on a logic analyser and the duty cycle is all over the place and won't give a stable enough signal to run the piece of equipment I need to run. 

See this post, seems to discuss the issue with the ESP-IDF 
https://www.esp32.com/viewtopic.php?t=8067

> with disabled "enable "Enable dynamic frequency scaling (DFS) at startup" it looks like this:

Seems disabling the dynamic frequency scaling at startup fixes this but I imagine that could create other issues.

neoniousTR, 2020-04-20:
OK. Interesting..

The board is not a neonious one? If this would have been the case, the PWMs would have been done by the LPC822 coprocessor.

If not, then you are on track that the LEDC interface of the ESP32 does the PWM. And the issue will be the one you linked.

Just to make sure about the priority: What do you need the PWM feature for?

Thank you!

Thomas

Craigzyc, 2020-04-20:
No priority simply letting you know about this undesired behaviour. This is not on the Neonious One.

I am taking analog signals from a build controller and converting them to other protocols. 2 are rs485, another is PWM. I just wrote a quick arduino for the PWM for now and have it running beside the ESP32. If this gets fixed I will likely rewire it back to the ESP. 
I may swap out for a Neonious One as well for the linear analog