neonious/lowjs #113: package.json "main" should redirect index.js

algiuxass, 2020-07-09:
Some node modules have in package.json "main" parameter which basically says which script to execute.

For example, people often put the entry point to lib/index.js or lib/<packagename>.js, in this case the corresponding script must be described as main in package.json.

Example of package.json:
```js
{
  "license": "BSD-3-Clause",
  "main": "src/index.js",
  "name": "packageName",
  "repository": {
    "type": "git",
    "url": "git://github.com/example/example.git"
  },
  "scripts": {
    "fix": "standard --fix",
    "lint": "standard",
    "prepare": "npm install require-self && require-self",
    "test": "mocha --recursive --reporter spec --exit"
  },
  "version": "1.13.0"
}
```

neoniousTR, 2020-07-13:
OK. But what exactly is wrong with low.js regarding this? Where is the bug?

algiuxass, 2020-07-14:
> OK. But what exactly is wrong with low.js regarding this? Where is the bug?

Low.js does not read package.json, that means it won't work on some packages which doesn't have "index.js". Well, people could manually write index.js files to route it to correct location/file but this is a hacky solution.

neoniousTR, 2020-07-19:
It does read package.json, see src/low_module.cpp. Do you have an exact example where this does not work?

closed 2021-02-21 by ThomasRogg