neonious/lowjs #114: require json is not working

algiuxass, 2020-07-09:
Example from avj node module:
```js
var metaSchema = require('./refs/json-schema-draft-07.json');
```
```
TypeError: cannot resolve module './refs/json-schema-draft-07.json', parent '..../node_modules/ajv/lib/definition_schema.js'
```

Make it possible to require json files.

neoniousTR, 2020-07-13:
low.js should support requiring JSON files. I will try myself next weekend and report!

neoniousTR, 2020-07-19:
Requiring JSON files works perfectly. I guess the problem is the "..../" instead of a "../../", so the resolving has some kind of bug. Do you have an example project or steps to do to trigger such an error?

algiuxass, 2020-07-29:
Sorry, forgot to mention the "...." was to hide full path or whatever was it. It's shouldn't be a problem now. Thanks!

closed 2020-09-27 by algiuxass