neonious/lowjs #118: tensorflow/tfjs-node support

MyIsaak, 2020-08-09:
I would like to see Tensorflow JS supported on this platform. Currently they have support in a flashed C++ backend for some microcontrollers (ESP32, Arduino Mini) see here https://www.tensorflow.org/lite/microcontrollers/get_started

My current solution is going for a larger microcontroller with a full linux (Arduino), then installing `node` with `@tensorflow/tfjs-node` via npm. 

Also note that Tensorflow installs a precompiled ARM32/64 backend, which gets linked via node-gyp. So does low.js support such an installation?

neoniousTR, 2020-08-12:
TensorFlow is fun with my GeForce 2080 RTX Ti, but with a microcontroller? The use cases are very limited, so I do not think many people would be interested in TensorFlow in low.js..

But if you would like to see it, maybe you would like to integrate both? I am always open to pull requests :-)

MyIsaak, 2020-08-13:
It's interesting for those in the gadget market who want to create cheaper products. With some careful optimizations on model size and inference time (e.g. post weight optimisation, and byte size reduction) it is possible to sacrifice accuracy for faster inference time.

In a few weeks (when I find my lost ESP8266), I'll test this out on low.js.

closed 2021-02-21 by ThomasRogg